lgsvl/simulator #1: Document?

poutyface, 2018-09-28:
I watch Apollo demo on youtube.
I'd like to try it, Are there any documents(install, usage etc..)?

zelenkovsky, 2018-09-28:
Hello, thanks for your request.

Yes you can try running Apollo with the simulator even thought it is still
under development.

Let me put you in touch with Andrey K. He will help you to get it up and
running.

We are going to publish more information about initial setup as well.

PS: you can build and run simulator without Apollo as well. Just use latest
Unity3D to build it and start, then press shift+RUN button after selecting
map and vehicle.

BR
Dmitry Z.

poutyface, 2018-10-01:
Thank you for your reply. 
Using docker compose, I can run simulator. 

Then, All I have to do is to check Apollo's ROS message 

Great work, thanks.

closed 2018-10-01 by poutyface