lgsvl/simulator #2: failed to build docker.

hakuturu583, 2018-10-01:
Yesterday, I try to build docker image like this.
```
cd Docker
sudo docker build -t lg_sim .
```

Then, the build started but it suddenly stop and show message like below.
[Image](https://pbs.twimg.com/media/DoVYVbHVsAEXFB6.jpg:large)

I think this is the license agreement of Unity.
And then, I cannot type y in the terminal.
So, I could not agree licence of Unity and build the simulator.

Please help!!

martins-mozeiko, 2018-10-02:
It is OK to see license agreement for a long time, the build of docker image is still happening in background, the unity installer executable is simply buffering stdout, so it is not immediately visible what is happening. The image build should finish in 5-10 minutes depending on your internet speed (it downloads whole Unity installer).

That said, the files in docker folder are currently used in our internal Jenkins server and they require a bit of extra setup to be used. Which is currently not documented in this repo. I recommend not using them.

Please use following command-line on your host to build standalone player of simulator on Windows:
```
/path/to/unit/installation/Unity \
    -batchmode \
    -nographics \
    -silent-crashes \
    -quit \
    -buildDestination /PATH/TO/OUTPUT/FOLDER/simulator.exe \
    -buildTarget Win64 \
    -executeMethod BuildScript.Build \
    -projectPath /PATH/TO/GIT/CHECKOUT
```

And use following command to build on Linux:
```
/path/to/unit/installation/Unity \
    -batchmode \
    -nographics \
    -silent-crashes \
    -quit \
    -buildDestination /PATH/TO/OUTPUT/FOLDER/simulator \
    -buildTarget Linux64 \
    -executeMethod BuildScript.Build \
    -projectPath /PATH/TO/GIT/CHECKOUT
```

andytolst, 2018-10-08:
@hakuturu583 Please let us know if you are able to make the build or if you are still having any issues. Thanks.

closed 2018-10-18 by andytolst