lgsvl/simulator #3: How to connect ROS.

hakuturu583, 2018-10-01:
I try ROS bridge with your sim.
I saw left top of your sim shows that your sim is connecting to ROS, but, I cannot find any topics in my ROS network.

[image](https://pbs.twimg.com/media/DobE0oNUYAY0HOp.jpg:large)

andytolst, 2018-10-01:
Please try with the following environment and robot:
"DuckieDowntown" as map -- "Duckiebot-duckietown-ros1" as robot
you should be able to see the image from topic "/simulator/camera_node/image/compressed"

hakuturu583, 2018-10-02:
Thanks !! I succeed to build connection your sim between ROS.

closed 2018-10-02 by hakuturu583

ErijBNT, 2019-05-06:
I did that but I have got this error.
How to fix that?
[image]

martins-mozeiko, 2019-05-06:
@ErijBNT Do you have duckietown_msgs and autoware_msgs packages installed in your ROS environment?

ErijBNT, 2019-05-06:
@martins-mozeiko No, I don't.
why is it necessary to have them installed?

martins-mozeiko, 2019-05-06:
If you want to control the duckiebot inside simulator, it will listen for `duckietown_msgs/WheelsCmdStamped` message type that your ROS node should publish. If you don't have such message, you won't be able to control it from ROS.

Similarly for autoware_msgs - to control duckiebot from autoware it listens for `autoware_msgs/VehicleCmd` message type. If you want to control it from Autoware, you need this message type.

If you don't need to control duckiebot from ROS, then you can ignore these errors. Everything else will still work.

ErijBNT, 2019-05-07:
@martins-mozeiko Thank you for your response! I was only trying to test that. 
But, in fact, I want to create a car with some sensors and to control it using ros. I won't need autoware or anything else. Do you know how can I do that in LG simulator? 
You will help me a lot.

martins-mozeiko, 2019-05-08:
Check out our [lanefollowing](https://github.com/lgsvl/lanefollowing) project. It shows how to create simple "AD stack" for controlling car based only on camera. In this repository you'll find code to receive camera image from car, process it with tensorflow, and send back control command to simulator for car to stay in middle of lanes.

This repository uses vanilla ROS2, no Apollo, no Autoware.

ErijBNT, 2019-05-08:
Thank you for your help!

nhatuan84, 2021-07-21:
@hadiTab my Simulator is on another Machine, Autoware is on Xavier. Do you know where can I config the IP address. Or how to connect them?

Thanks.