lgsvl/simulator #5: No custom message.

hakuturu583, 2018-10-02:
I connected your sim to Autoware, but I found such problem.

```
[ERROR] [1538477393.150159]: [Client 0] advertise: Unable to load the manifest for package duckietown_msgs. Caused by: duckietown_msgs
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/masaya/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
```

Is this your custom message isn't it??
Please open this custom message package.

hakuturu583, 2018-10-03:
Thanks I succeed.

closed 2018-10-03 by hakuturu583

thanh118, 2018-10-17:
Hello, how about this error: I got this error directly by autoware_msgs:

[INFO] [WallTime: 1539801934.433567] Client connected.  1 clients total.
[ERROR] [WallTime: 1539801947.689685] [Client 12] subscribe: Unable to load the manifest for package autoware_msgs. Caused by: autoware_msgs

andytolst, 2018-10-17:
@thanh118 are you running Autoware and rosbridge in the same ROS environment ? i.e.  run
```
source ./devel/setup.bash
```
before running rosbridge

andytolst, 2018-10-17:
@thanh118 I don't see where is your Autoware located. Please make sure you have 
Autoware/ros/src/msgs/autoware_msgs/ and all message descriptions are there, e.g. can you do
```
cat  Autoware/ros/src/msgs/autoware_msgs/msg/VehicleCmd.msg
```

thanh118, 2018-10-18:
Hi andy, 
it here:
thanh@thanh-Inspiron-15-7000-Gaming:~$ cat Autoware/ros/src/msgs/autoware_msgs/msg/VehicleCmd.msg
Header header
autoware_msgs/steer_cmd steer_cmd
autoware_msgs/accel_cmd accel_cmd
autoware_msgs/brake_cmd brake_cmd
autoware_msgs/lamp_cmd lamp_cmd
int32 gear
int32 mode
geometry_msgs/TwistStamped twist_cmd
autoware_msgs/ControlCommand ctrl_cmd
int32 emergency

andytolst, 2018-10-18:
then you need to source it
```
source /home/thanh/Autoware/ros/devel/setup.bash
```