lgsvl/simulator #6: Incorrect rendering region of depth

atinfinity, 2018-10-03:
## System information (version)
- Simulator =>  [auto-simulator-linux64-474.zip](https://media.githubusercontent.com/media/lgsvl/simulator/binaries/auto-simulator-linux64-474.zip)
- Operating System / Platform => Ubuntu 16.04 64 bit

## Detailed description
I tried "Enable Color Segment Camera" mode in this simulator.  
As a result, I think that the display of "Segmentation" is no problem.

[image]

But, I think the rendering region of "depth" is not appropriate.

[image]

## Steps to reproduce
1. Launch simulator
1. Select **1920 x 1080** as resolution
2. Select **Free Roaming** mode
3. Select **SanFrancisco** map and **XE_Rigged-apollo** robot
4. Select **RUN**
5. Check **Enable Color Segment Camera**

andytolst, 2018-10-03:
I guess this is a know issue, just press 'S' or 'D' key multiple times, and it will re-render in  a correct size.
We'll try to fix it in future releases

atinfinity, 2018-10-04:
Thank you for your reply! I'm looking forward to resolve in future releases.

hadiTab, 2018-10-04:
@atinfinity this was just fixed with the latest commit.

andytolst, 2018-10-08:
@atinfinity Could you check the latest code from master and see if the problem is fixed. Thanks

andytolst, 2018-10-18:
if you feel that there are still some issues with segment camera - feel free to reopen

closed 2018-10-18 by andytolst