lgsvl/simulator #7: Build error with docker-compose

rongguodong, 2018-10-03:
When I run "docker-compose build", I got the following error messages. How can I fix it?

ERROR: Version in "./docker-compose.yml" is unsupported. You might be seeing this error because you're using the wrong Compose file version. Either specify a version of "2" (or "2.0") and place your service definitions under the `services` key, or omit the `version` key and place your service definitions at the root of the file to use version 1.
For more on the Compose file format versions, see https://docs.docker.com/compose/compose-file/

andytolst, 2018-10-03:
Please try to update docker-compose: https://docs.docker.com/compose/install/#install-compose

```
sudo curl -L "https://github.com/docker/compose/releases/download/1.22.0/docker-compose-$(uname -s)-$(uname -m)" -o /usr/local/bin/docker-compose

sudo chmod +x /usr/local/bin/docker-compose

/usr/local/bin/docker-compose build
```

andytolst, 2018-10-08:
@rongguodong Are you able to build the simulator with the updated docker-compose ?

rongguodong, 2018-10-11:
I now got errors for apt-get update:

==============================================
~/lgsvl/simulator/Docker$ /usr/local/bin/docker-compose build
WARNING: The UID variable is not set. Defaulting to a blank string.
Building auto-sim-build
Step 1/2 : FROM ubuntu:16.04
 ---> b9e15a5d1e1a
Step 2/2 : RUN set -ex     && apt-get update     && DEBIAN_FRONTEND=noninteractive apt-get upgrade -y     && DEBIAN_FRONTEND=noninteractive apt-get install -y         sudo curl xvfb zip         gconf-service lib32gcc1 lib32stdc++6 libasound2 libc6 libc6-i386 libcairo2 libcap2 libcups2         libdbus-1-3 libexpat1 libfontconfig1 libfreetype6 libgcc1 libgconf-2-4 libgdk-pixbuf2.0-0         libgl1 libglib2.0-0 libglu1 libgtk2.0-0 libnspr4 libnss3 libpango1.0-0 libstdc++6 libx11-6         libxcomposite1 libxcursor1 libxdamage1 libxext6 libxfixes3 libxi6 libxrandr2 libxrender1         libxtst6 zlib1g debconf npm libgtk2.0-0 libsoup2.4-1 libarchive13 libpng16-16     && apt-get clean     && curl -fLo UnitySetup https://beta.unity3d.com/download/170f0691b973/UnitySetup-2018.1.0f2     && chmod +x UnitySetup     && yes | ./UnitySetup --unattended --install-location=/opt/Unity --components=Unity,Windows-Mono     && rm UnitySetup
 ---> Running in 7b5b4a4e2e17
+ apt-get update
Err:1 http://archive.ubuntu.com/ubuntu xenial InRelease
  Temporary failure resolving 'archive.ubuntu.com'
Err:2 http://security.ubuntu.com/ubuntu xenial-security InRelease
  Temporary failure resolving 'security.ubuntu.com'
Err:3 http://archive.ubuntu.com/ubuntu xenial-updates InRelease
  Temporary failure resolving 'archive.ubuntu.com'
Err:4 http://archive.ubuntu.com/ubuntu xenial-backports InRelease
  Temporary failure resolving 'archive.ubuntu.com'
Reading package lists...
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial/InRelease  Temporary failure resolving 'archive.ubuntu.com'
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial-updates/InRelease  Temporary failure resolving 'archive.ubuntu.com'
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial-backports/InRelease  Temporary failure resolving 'archive.ubuntu.com'
W: Failed to fetch http://security.ubuntu.com/ubuntu/dists/xenial-security/InRelease  Temporary failure resolving 'security.ubuntu.com'
W: Some index files failed to download. They have been ignored, or old ones used instead.
+ DEBIAN_FRONTEND=noninteractive apt-get upgrade -y
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ DEBIAN_FRONTEND=noninteractive apt-get install -y sudo curl xvfb zip gconf-service lib32gcc1 lib32stdc++6 libasound2 libc6 libc6-i386 libcairo2 libcap2 libcups2 libdbus-1-3 libexpat1 libfontconfig1 libfreetype6 libgcc1 libgconf-2-4 libgdk-pixbuf2.0-0 libgl1 libglib2.0-0 libglu1 libgtk2.0-0 libnspr4 libnss3 libpango1.0-0 libstdc++6 libx11-6 libxcomposite1 libxcursor1 libxdamage1 libxext6 libxfixes3 libxi6 libxrandr2 libxrender1 libxtst6 zlib1g debconf npm libgtk2.0-0 libsoup2.4-1 libarchive13 libpng16-16
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package sudo
E: Unable to locate package curl
E: Unable to locate package xvfb
E: Unable to locate package zip
E: Unable to locate package gconf-service
E: Unable to locate package lib32gcc1
E: Unable to locate package lib32stdc++6
E: Couldn't find any package by regex 'lib32stdc++6'
E: Unable to locate package libasound2
E: Unable to locate package libc6-i386
E: Unable to locate package libcairo2
E: Unable to locate package libcups2
E: Unable to locate package libdbus-1-3
E: Unable to locate package libexpat1
E: Unable to locate package libfontconfig1
E: Unable to locate package libfreetype6
E: Unable to locate package libgconf-2-4
E: Unable to locate package libgdk-pixbuf2.0-0
E: Couldn't find any package by glob 'libgdk-pixbuf2.0-0'
E: Couldn't find any package by regex 'libgdk-pixbuf2.0-0'
E: Unable to locate package libgl1
E: Unable to locate package libglib2.0-0
E: Couldn't find any package by glob 'libglib2.0-0'
E: Couldn't find any package by regex 'libglib2.0-0'
E: Unable to locate package libglu1
E: Unable to locate package libgtk2.0-0
E: Couldn't find any package by glob 'libgtk2.0-0'
E: Couldn't find any package by regex 'libgtk2.0-0'
E: Unable to locate package libnspr4
E: Unable to locate package libnss3
E: Unable to locate package libpango1.0-0
E: Couldn't find any package by glob 'libpango1.0-0'
E: Couldn't find any package by regex 'libpango1.0-0'
E: Unable to locate package libx11-6
E: Unable to locate package libxcomposite1
E: Unable to locate package libxcursor1
E: Unable to locate package libxdamage1
E: Unable to locate package libxext6
E: Unable to locate package libxfixes3
E: Unable to locate package libxi6
E: Unable to locate package libxrandr2
E: Unable to locate package libxrender1
E: Unable to locate package libxtst6
E: Unable to locate package npm
E: Unable to locate package libgtk2.0-0
E: Couldn't find any package by glob 'libgtk2.0-0'
E: Couldn't find any package by regex 'libgtk2.0-0'
E: Unable to locate package libsoup2.4-1
E: Couldn't find any package by glob 'libsoup2.4-1'
E: Couldn't find any package by regex 'libsoup2.4-1'
E: Unable to locate package libarchive13
E: Unable to locate package libpng16-16
[... cut ...] libstdc++6 libx11-6         libxcomposite1 libxcursor1 libxdamage1 libxext6 libxfixes3 libxi6 libxrandr2 libxrender1         libxtst6 zlib1g debconf npm libgtk2.0-0 libsoup2.4-1 libarchive13 libpng16-16     && apt-get clean     && curl -fLo UnitySetup https://beta.unity3d.com/download/170f0691b973/UnitySetup-2018.1.0f2     && chmod +x UnitySetup     && yes | ./UnitySetup --unattended --install-location=/opt/Unity --components=Unity,Windows-Mono     && rm UnitySetup' returned a non-zero code: 100

andytolst, 2018-10-11:
this looks like a DNS problem in docker. are you able to use 8.8.8.8 DNS in your network? 
> nslookup google.com 8.8.8.8

if so, you can try to add that in /etc/default/docker
DOCKER_OPTS="--dns 8.8.8.8 --dns 8.8.4.4"

otherwise you can look for solution here:
https://stackoverflow.com/questions/24991136/docker-build-could-not-resolve-archive-ubuntu-com-apt-get-fails-to-install-a

andytolst, 2018-10-15:
@rongguodong 
If you still have issues with Docker, you can try to build it locally on you host machine:
```
/opt/Unity/Editor/Unity \
    -batchmode \
    -nographics \
    -silent-crashes \
    -quit \
    -buildDestination /path/to/output/folder/simulator \
    -buildTarget Linux64 \
    -executeMethod BuildScript.Build \
    -projectPath /path/to/simulator/source/code

```

closed 2018-10-20 by andytolst

EricBoiseLGSVL, 2020-09-29:
@rongguodong lol this is you using simulator way back?

rongguodong, 2020-09-29:
> @rongguodong lol this is you using simulator way back?

I do not even remember what I was talking about. :p