lgsvl/simulator #10: license error with xvfb-run

rguptatx, 2018-10-05:
on executing xvfb-run
/usr/bin/xvfb-run /opt/Unity/Editor/Unity \
>     -batchmode \
>     -nographics \
>     -silent-crashes \
>     -quit \
>     -buildDestination ./build/simulator \
>     -buildTarget Linux64 \
>     -executeMethod BuildScript.Build \
>     -projectPath . \
>     -logFile /dev/stdout

I get the error:
Cancelling DisplayDialog: Failed to activate/update license. Missing or bad username and password. Please try again using valid credentials or contact support@unity3d.com
This should not be called in batch mode.

andytolst, 2018-10-06:
Have you activated Unity ? are you able to use Editor and open simulator project in the Editor ?

rguptatx, 2018-10-06:
No, I got a personal license for Unity and have not been able to figure out how to activate it and link it to this simulator. Any pointers will be appreciated.

andytolst, 2018-10-06:
https://docs.unity3d.com/Manual/OnlineActivationGuide.html

simply run the Editor  /opt/Unity/Editor/Unity and follow the steps

chris490yy, 2018-10-12:
> https://docs.unity3d.com/Manual/OnlineActivationGuide.html
> 
> simply run the Editor /opt/Unity/Editor/Unity and follow the steps

HI

When I run the Editor /opt/Unity/Editor/Unity and encounter the login window, my keyboard is disabled for typing any characters to login or create an account. I'm using the ubuntu 16.04. Have you met with the same problem?

Thanks

andytolst, 2018-10-12:
we haven't seen this problem with Unity Editor, can you try to re-install it ? 
please make sure you are using version 2018.2.4f1

chris490yy, 2018-10-16:
> we haven't seen this problem with Unity Editor, can you try to re-install it ?
> please make sure you are using version 2018.2.4f1

Thanks, it works now. 

Is there any instruction for building my own mobile robot with specific sensor combinations, like IMU, stereo cameras?

andytolst, 2018-10-16:
we don't have specific instructions, but you can take a look at Assets/Scripts/ImuSensor.cs and existing camera. 
you can also try modifying existing Duckiebot robot Assets/Prefabs/RobotCandidates/duckiebot-duckietown-ros1.prefab

closed 2018-10-18 by andytolst