lgsvl/simulator #11: Compiled the project on Ubuntu 16.04I am unable to connect to rosbridge

bhaskarsuper9000, 2018-10-08:


andytolst, 2018-10-08:
@bhaskarsuper9000 can we get more details, do you see any errors in rosbridge, do you see any clients connected in the bridge, IP address and port numbers, etc.

bhaskarsuper9000, 2018-10-08:
rosbridge running on its default on port 9090

andytolst, 2018-10-08:
let's try one more thing: start rosbridge in console, start simulator, select "Free Roaming", 
select "DuckieDowntown" as map and "duckiebot-duckietown-ros1" as robot.
You should see something like this:
[image]

bhaskarsuper9000, 2018-10-08:
[image]
Allright, I tried it, but I still see connecting. Don't think its a ROS issue, because although I have multiple versions of ROS, I am using kinetic for this one. (sourced the correct one)

I checked the log /home/$user/.config/unity3d/LGE/Duckietown/Player.log

```
NullReferenceException: Object reference not set to an instance of an object
  at VideoToROS.Update () [0x00000] in <9a5e55b624b8451287983db3d8d64cfd>:0 
 
(Filename: <9a5e55b624b8451287983db3d8d64cfd> Line: 0)

10/8/2018 12:53:28 PM|Fatal|WebSocket.connect|An exception has occurred while reading an HTTP request/response.
An exception has occurred while attempting to connect.
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 43)

NullReferenceException: Object reference not set to an instance of an object
  at VideoToROS.Update () [0x00000] in <9a5e55b624b8451287983db3d8d64cfd>:0 
 
(Filename: <9a5e55b624b8451287983db3d8d64cfd> Line: 0)

NullReferenceException: Object reference not set to an instance of an object
  at VideoToROS.Update () [0x00000] in <9a5e55b624b8451287983db3d8d64cfd>:0 
 
(Filename: <9a5e55b624b8451287983db3d8d64cfd> Line: 0)

10/8/2018 12:53:28 PM|Fatal|WebSocket.connect|An exception has occurred while reading an HTTP request/response.
An exception has occurred while attempting to connect.
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 43)

NullReferenceException: Object reference not set to an instance of an object
  at VideoToROS.Update () [0x00000] in <9a5e55b624b8451287983db3d8d64cfd>:0 
 
(Filename: <9a5e55b624b8451287983db3d8d64cfd> Line: 0)

NullReferenceException: Object reference not set to an instance of an object
  at VideoToROS.Update () [0x00000] in <9a5e55b624b8451287983db3d8d64cfd>:0 
 
(Filename: <9a5e55b624b8451287983db3d8d64cfd> Line: 0)
```
Does this give a clue on whats going on?

bhaskarsuper9000, 2018-10-08:
Okay, we are past this. I got it to connected. It seems I had an issue with rosbridge, it shows connected for now. And I am able to get the data with rostopic echo. Thanks, do you have any suggestion for using a visualizer?

hadiTab, 2018-10-08:
@bhaskarsuper9000  You should be able to visualize most of the messages in Rviz. Are there any specific messages you are having difficulty with?

bhaskarsuper9000, 2018-10-08:
Thanks @hadiTab and @andytolst, I was able to visualize the lidar and camera messages with rviz. I have some experience with the apollo stack as well. I want to try that next, will close this issue for now.

closed 2018-10-08 by bhaskarsuper9000

kk2491, 2018-11-02:
@bhaskarsuper9000 Could you please help me in fixing the below error:

    roslaunch rosbridge_server rosbridge_websocket.launch
    RLException: [rosbridge_websocket.launch] is neither a launch file in package [rosbridge_server] nor     is [rosbridge_server] a launch file name
    The traceback for the exception was written to the log file

Do I need to install rosbridge separately ?

Thank you,
KK

martins-mozeiko, 2018-11-02:
@kk2491 is this for Apollo or for Autoware?

If you are using Apollo, make sure you have cloned git repository together with submodules (`--recursive` argument). If you have already cloned it, you can fix submodules by running `init` and `update` submoduke commands.

hadiTab, 2018-11-02:
@kk2491 rosbridge is included as a git submodule in the repository. Make sure you use the `--recurse-submodules` flag when cloning. It should be located in `apollo/ros_pkgs/src/rosbridge_suite`.
You will also need to build it in docker to use it, which can be done by running `catkin_make` from the container after changing directory to `apollo/ros_pkgs`.

If you have done all of these and you are still running into an issue, it is likely because the catkin_workspace has not been sourced.
I suggest running the rosbridge using the script we have provided in `apollo/scripts/rosbridge.sh`.

If you wish to manually launch it you can source the `setup.bash` file in `ros_pkgs`.

bhaskarsuper9000, 2018-11-02:
The above answer is perfect if you are running it with apollo. But in case you are just using a standard ROS repository like indigo or kinetic with Ubuntu, you can also just use 
`sudo apt-get install ros-kinetic-rosbridge-*` to install the rosbridge related packages

kk2491, 2018-11-05:
@hadiTab Thanks for the details, let me try and let you know the status.

kk2491, 2018-11-05:
@bhaskarsuper9000 Thank you for the answer. Currently I am running LG simulator standalone. I have ROS version Lunar installed in my system. Unfortunately I am not able to install rosbridge by running ```sudo apt-get install ros-kinetic-rosbridge-*```. 

Seems like it is not supported in Lunar. Has anybody faced this issue?

Thank you,
KK

hadiTab, 2018-11-05:
@kk2491  rosbridge is supported in Lunar. You can install it using:

`sudo apt-get install ros-lunar-rosbridge-server`

Here's a [link](http://wiki.ros.org/rosbridge_suite) to the ros wiki page for rosbrige. The link

trusktr, 2021-04-02:
> you can also just use
> `sudo apt-get install ros-kinetic-rosbridge-*` to install the rosbridge related packages

I have that stuff installed, but I still get the same error `[rosbridge_websocket.launch] is neither a launch file...`.

F.e. I see this:

```sh
vagrant@ubuntu-bionic:~$ apt list ros-melodic-rosbridge-server 
Listing... Done
ros-melodic-rosbridge-server/bionic 0.11.13-1bionic.20210112.195018 amd64

vagrant@ubuntu-bionic:~$ roslaunch rosbridge_server rosbridge_websocket.launch
RLException: [rosbridge_websocket.launch] is neither a launch file in package [rosbridge_server] nor is [rosbridge_server] a launch file name
The traceback for the exception was written to the log file
```

I read the [RunningRosbridge](http://wiki.ros.org/rosbridge_suite/Tutorials/RunningRosbridge) and it seems like it should just work.

Any ideas what to check for?