lgsvl/simulator #13: how to build the entire project

ImCabbage, 2018-10-14:
Hi, 
I follow the step and successfully build the image (docker-compose build) and I think the next step is "docker-compose up" so that I can directly get the built files. Am I right?
But it seems I got "auto-sim-build exited with code 0" when I tried to run "docker-compose up".

BTW, I use the autoware commit.

Thanks.

martins-mozeiko, 2018-10-15:
Docker in our repository is used only to automate building on our Jenkins CI infrastructure.

I suggest you to build our Simulator without docker. As long as your have Unity installed & activated on your desktop, the simulator should build just fine.

On Linux you can build it with following command:
```
/opt/Unity/Editor/Unity \
    -batchmode \
    -nographics \
    -silent-crashes \
    -quit \
    -buildDestination /path/to/output/folder/simulator \
    -buildTarget Linux64 \
    -executeMethod BuildScript.Build \
    -projectPath /path/to/simulator/source/code
```

If you want to use docker, we don't use docker-compose to run services (up/down commands). We use it to run activate or build shell scripts. First you'll need to activate Unity that is inside docker with following command:
```
docker-compose run auto-sim-activate
```

Then you should be able to run build with following two commands:
```
export UID
docker-compose run auto-sim-build
```
export will make docker container to run under your user, not as root.

The output should be two zip files - one for Windows, one for Linux build.

Let me know if you have more questions on this.

ImCabbage, 2018-10-17:
@martins-mozeiko 
Thanks for your reply. It really helps me. But I still met a build problem.
When I tried to build the entire project using Unity, I got:

executeMethod class 'BuildScript' could not be found.
Argument was -executeMethod BuildScript.Build
 
(Filename: /home/builduser/buildslave/unity/build/Runtime/Utilities/Argv.cpp Line: 295)

Aborting batchmode due to failure:
executeMethod class 'BuildScript' could not be found.
Argument was -executeMethod BuildScript.Build

I think there is something missing, am I right? I will appreciate it if you can help.

andytolst, 2018-10-17:
please make sure that '-projectPath /path/to/simulator/source/code' is pointing to your simulator source folder
you can also check that this file exists: Assets/Scripts/Editor/BuildScript.cs

ImCabbage, 2018-10-18:
@andytolst 
Thanks for your reply. Unfortunately, I still cannot build it successfully. I am pretty sure that I put the right address and Assets/Scripts/Editor/BuildScript.cs stays where it belongs. I still got the following error:

Assets/Scripts/Ros/Bridge.cs(12,7): error CS0246: The type or namespace name `WebSocketSharp' could not be found. Are you missing an assembly reference?
 
(Filename: Assets/Scripts/Ros/Bridge.cs Line: 12)

Assets/UnityPackages/JsonNetSample/JsonNetSample.cs(5,7): error CS0246: The type or namespace name `Newtonsoft' could not be found. Are you missing an assembly reference?
 
(Filename: Assets/UnityPackages/JsonNetSample/JsonNetSample.cs Line: 5)

Assets/UnityPackages/JsonNetSample/JsonNetSample.cs(6,7): error CS0246: The type or namespace name `Newtonsoft' could not be found. Are you missing an assembly reference?
 
(Filename: Assets/UnityPackages/JsonNetSample/JsonNetSample.cs Line: 6)

Assets/Scripts/Ros/Bridge.cs(38,9): error CS0246: The type or namespace name `WebSocket' could not be found. Are you missing `System.Net.WebSockets' using directive?
 
(Filename: Assets/Scripts/Ros/Bridge.cs Line: 38)

Assets/Scripts/Ros/Bridge.cs(108,37): error CS0246: The type or namespace name `CloseEventArgs' could not be found. Are you missing an assembly reference?
 
(Filename: Assets/Scripts/Ros/Bridge.cs Line: 108)

Assets/Scripts/Ros/Bridge.cs(122,37): error CS0246: The type or namespace name `ErrorEventArgs' could not be found. Are you missing `System.IO' using directive?
 
(Filename: Assets/Scripts/Ros/Bridge.cs Line: 122)

Assets/Scripts/Ros/Bridge.cs(141,39): error CS0246: The type or namespace name `MessageEventArgs' could not be found. Are you missing `UnityEngine.Networking.PlayerConnection' using directive?
 
(Filename: Assets/Scripts/Ros/Bridge.cs Line: 141)

executeMethod class 'BuildScript' could not be found.
Argument was -executeMethod BuildScript.Build
 
(Filename: /home/builduser/buildslave/unity/build/Runtime/Utilities/Argv.cpp Line: 295)

Are there any other dependency you have for this project. I think there is something important missing.

andytolst, 2018-10-18:
could you check that you have the following dlls in your simulator source:
```
ls Assets/Scripts/Ros/websocket-sharp.dll
Assets/Scripts/Ros/websocket-sharp.dll

ls Assets/UnityPackages/JsonNet/Newtonsoft.Json.dll
Assets/UnityPackages/JsonNet/Newtonsoft.Json.dll
```

andytolst, 2018-10-18:
please make sure you have pulled all lfs objects:
```
git lfs pull
git lfs checkout
```

ImCabbage, 2018-10-19:
@andytolst 
Thanks for your help. It works. I think you should put these two lines in README.md.
Previously, I saw they were there. But somehow in the latest version, you deleted them. Just a suggestion.

andytolst, 2018-10-19:
@ImCabbage it should work by default without these commands if GIT LFS is installed properly, see https://github.com/lgsvl/simulator/pull/18

closed 2018-10-20 by andytolst