lgsvl/simulator #14: how to config the sensor and vehicle

gogojjh, 2018-10-17:
As the title says, how can I set a different vehicle with different sensors in your simulator? LIke I want to design a car/bus with lidars and cameras at different positions.
Thanks!

gogojjh, 2018-10-18:
Yes, based on Apollo. I find a problem with the lidar data, Like the below figure, there exists a distortion of the point cloud when the car is moving. How I fix this problem? Increase the frequency of lidar?
Thanks!
[image]

andytolst, 2018-10-18:
in general our point cloud is perfectly compensated for the motion of the car, i.e. points are exactly where they are in the real world, and this is what Apollo is doing in a special subnode in the Lidar driver (see "compensated" topic vs raw point cloud).

"sector" distortions you see are happening because points are collected in batches between the frames and these are visible on the high speeds when objects move faster between the frames. 
We are trying to improve our Lidar model in future releases to be more "smooth", meanwhile you also can try to play with Lidar parameters like resolution, number of rays of frequency. Disabling other sensors and decreasing resolution (to increase FPS) could also help.

gogojjh, 2018-10-19:
Sorry that I have another question. You mentioned that we can modify the current car/sensor setup in "Prefabs/RobotCandidates/XE_Rigged-apollo.prefab". Here I tried to modify the number of rays and frequency, but it did not change in the simulation. So there are any other steps I should take? 
[image]

Thanks!
Sorry again that I have too many problems TAT

gogojjh, 2018-10-19:
I just follow these steps and create a new prefab in the "RobotCandidates" fold. But when I run the simulator again, I still can not load the new robot. What can I do next? 
Thanks!

martins-mozeiko, 2018-10-19:
By "can not load" you mean it does not appear in combobox? The combobox in main menu is populated from RosRobot object. Open MainMenu scene, select RosRobots game object, and in inspector extend "Robot Candidates" array.

gogojjh, 2018-10-22:
My problem is this:
1. I created a robot based on the "XE_Rigged-autoware" like this:
[image]
2. I saved this robot model in the "Assets/Prefabs/RobotCandidates" fold
[image]
3. I opened the simulator, SanFrancisco scene, the robot I created cannot be loaded (the model "XE_Rigged-jjiao_stareo_lidar" doesn't appear).
[image]

You mentioned "MainMenu scene" and "RosRobots game object", where can I find them? 
Thanks!

gogojjh, 2018-10-22:
Let me explain this problem a bit clearer, for example
1. I created a robot model based on autoware like this,
[image]
2. I saved this model in the "Assets/Prefabs/RobotCandidates"
[image]
3. When I run the simulator and select the SanFrancisco map, I could not find the model I created here.
[image]
**So what additional steps should I take?
Moreover, you mentioned "RosRobot object" and "MainMenu scene", where can I find them?**
I hope these problems won't cost you much time ...
Thanks!

gogojjh, 2018-10-22:
> By "can not load" you mean it does not appear in combobox? The combobox in main menu is populated from RosRobot object. Open MainMenu scene, select RosRobots game object, and in inspector extend "Robot Candidates" array.

gogojjh, 2018-10-22:
Let me explain this problem a bit clearer, for example
1. I created a robot model based on autoware.
2. I saved this model in the "Assets/Prefabs/RobotCandidates"
[image]
3. When I run the simulator and select the SanFrancisco map, I could not find the model I created here.
[image]
**So what additional steps should I take?
Moreover, you mentioned "RosRobot object" and "MainMenu scene", where can I find them?**
I hope these problems won't cost you much time ...
Thanks!

martins-mozeiko, 2018-10-22:
See following screenshot.

[image]

1. open MainMenu scene (double-click)
2. select RosRobots game object (single-click)
3. edit RosRobot properties in inspector - add new entry by dragging your prefab from Project window (where you selected scene)

gogojjh, 2018-10-23:
It can work, thanks a lot!

closed 2018-10-23 by gogojjh