lgsvl/simulator #15: how to run simulator along with apollo

ImCabbage, 2018-10-17:
I am a developer working with Apollo recently. I would like to know is there any tutorial on how to run the Apollo and this simulator together.
1. How to deal with the map problem?
2. Is there any way to adjust the camera angle?
3. How to run them simultaneously?
4. How to run multiple cars?

I think it would be better if there is a tutorial on that. 
Thanks for your work. This project is interesting. :)

andytolst, 2018-10-17:
@ImCabbage we are in process of open-sourcing the instructions, stay tuned.

1. How to deal with the map problem?
we'll provide the map of our virtual San Francisco in the Apollo protobuf format. There is also a tool to annotate and export the map in the Unity project (see Assets/Scripts/HDMapTool.cs)
Point cloud maps for msf localization can be done using lidar - rosbag - scripts/msf_simple_map_creator.sh in Apollo

2. Is there any way to adjust the camera angle?
Look at Prefabs/RobotCandidates/XE_Rigged-apollo.prefab it has 2 cameras for Apollo:
CaptureCamera and TelephotoCamera

3. How to run them simultaneously?
We will provide the exact instructions and (most probaly) pre-built Apollo docker container

4. How to run multiple cars?
You can run other cars by ticking "Enable traffic" checkbox.
Running multiple ego cars (controlled by Apollo) is in our plan and will be supported in future

yangydavid, 2018-10-18:
Hi, 

What time are you guys planning release the Apollo's instructions? 

Thanks!

andytolst, 2018-10-18:
should be done by the end of next week

andytolst, 2018-10-20:
@ImCabbage  @yangydavid see Apollo instructions here:  https://github.com/lgsvl/apollo

ImCabbage, 2018-10-22:
@andytolst 
Thanks for your instruction. It is helpful and I have successfully run Apollo along with the simulator. But I met a localization problem. I found when the localization module is on, the position of the vehicle is totally different in the simulator and dreamviewer. Is there any other configuration in the localization module? I think I do not have to open the sim-control option in the dreamviewer, right?

andytolst, 2018-10-22:
@ImCabbage can you attach a screenshot of incorrect position ? do you see the map of SF in the dreamview ?

make sure localization is in RTK mode:
```
$ cat modules/localization/conf/localization_config.pb.txt 
localization_type: RTK
```
make sure you have localization module running in Apollo.
make sure you selected "San Francisco" map in dreamview
make sure you enabled IMU and GPS sensors in Simulator and rosbridge is running:
```
rostopic echo /apollo/sensor/gnss/best_pose

---
header {
  timestamp_sec: 1224022974.02
  sequence_num: 21578
}
measurement_time: 1224022974.02
sol_status: SOL_COMPUTED
sol_type: NARROW_INT
latitude: 37.7908574786
longitude: -122.399391029
height_msl: 0.0
undulation: 0.0
datum_id: WGS84
latitude_std_dev: 0.00999999977648
longitude_std_dev: 0.00999999977648
height_std_dev: 0.00999999977648
base_station_id: "0"
differential_age: 2.0
solution_age: 0.0
num_sats_tracked: 15
num_sats_in_solution: 15
num_sats_l1: 15
num_sats_multi: 12
extended_solution_status: 33
galileo_beidou_used_mask: 0
gps_glonass_used_mask: 51
```

ImCabbage, 2018-10-23:
@andytolst 
Thanks for your info. I think the problem is still unsolved. Let me explain my situation in detail. I may help solve the problem.

I follow the tutorial in your official Youtube channel along with the README file in the forked Apollo repo. According to my experience with the Apollo platform, I think I do it right and successfully get the topic "/apollo/sensor/gnss/best_pose", just like what you post above (RTK mode in localization for sure). But the map seems not loaded when I start the localization module (sim_control closed). What I saw is all black background. I got two feedback from the Dreamview panel. 

1. Hardware GPS triggers safety mode: No GNSS status message.
2. Failed to load default POI. Please make sure the file exists at /apollo/modules/map/data/san_francisco/default_end_way_point.txt

I am not sure whether the second info will affect. As I know, the default_end_way_point.txt file is generated through the lane.csv. But it seems that this file is indeed missing. 

When I open the sim_control option, I am able to see the map in the dreamview. I have a screenshot of what I met when this option is on, sadly, uploading pictures in issues is unavailable. If you could give a mail address or something, I think I can email you the pics. But still, the car is not in the right position. In dreamview, the vehicle always starts at the bottom right corner wherever it is in the simulator. I think it's the default start point for the map in Apollo. 

BTW, sometimes I got error info when I start the rosbridge launch file, recorded as below:
 
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
/use_sim_time is not set, will not subscribe to simulated time [/clock] topic
/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_server/scripts/rosbridge_websocket:218 <module>() [INFO] [1540261198.005100]: Rosbridge WebSocket server started on port 9090
/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_server/src/rosbridge_server/websocket_handler.py:77 open() [INFO] [1540261198.010214]: Client connected.  1 clients total.
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_library/src/rosbridge_library/protocol.py:383 log() [INFO] [1540261199.806602]: [Client 0] Subscribed to /apollo/control
**/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_library/src/rosbridge_library/protocol.py:379 log() [ERROR] [1540261200.237110]: [Client 0] publish: integer out of range for 'I' format code**

hadiTab, 2018-10-23:
@ImCabbage, the two feedback messages you are seeing in dreamview are ok. The GPS message will appear because Apollo is not able to communicate with the GPS driver (which is not used with the simulator). The default POI is related to routing but should not affect localization.

The sim_control mode should not be used with the simulator. That mode allows apollo to simulate driving at a high level on the HD map without any sensor data.

Since it seems like you are having issues with rosbridge, I suspect that the GPS messages are not going through which would cause localization issues.

To confirm, please try echoing the GNSS topic inside docker:
`rostopic echo /apollo/sensor/gnss/best_pose`

This should print the GNSS messages on the screen. I suspect that this will not happen.

Did you have any errors when building ros_pkgs?
I will look into the error message you have posted. In the meantime you can try rebuilding ros_pkgs to make sure it was built properly.

To rebuild, first delete the `build` and `devel` folders located in `/apollo/ros_pkgs/`.
Then (from within the docker container) run `catkin_make`. Make sure that there are no build errors.

andytolst, 2018-10-23:
@ImCabbage please make sure you are using the latest code of the simulator (we changed GPS origin a week ago and it has to be in sync with Apollo)
also make sure Apollo is in "Standard" mode, not in "Navigation"
please also post the exact contents of /apollo/sensor/gnss/best_pose so we can check latitude / longitude

andytolst, 2018-10-24:
@yangydavid 
https://github.com/ApolloAuto/apollo/issues/4388
https://github.com/ApolloAuto/apollo/issues/5344

yangydavid, 2018-10-24:
@andytolst one more question I have is that the simulator runs very slow, and I use `top` command and found that ~90% cpu were used, I'm wondering if there's any trick to boost it?

andytolst, 2018-10-24:
@yangydavid it depends on your HW config, good CPU and graphics card usually help
- try to reduce the resolution of the simulator (especially if you are running it on 4k monitor) and run it in 'windowed' mode
- disable Lidar if you don't need it (it consumes the most of CPU time, should be improved in future releases)
- if you are running on Linux - install the latest Nvidia drivers (I guess this is done already)
- if nothing helps - try to running it on separate machine with Windows

yangydavid, 2018-10-24:
@andytolst 
Thanks for your comments!

ImCabbage, 2018-10-25:
@andytolst @hadiTab 
Thanks for your help. My problem is finally solved when I clone the simulator repo again. I think @andytolst  mentioned it right.

hadiTab, 2018-10-25:
I'm closing this issue since it seems to be resolved for everyone. Feel free to reopen it if needed.

closed 2018-10-25 by hadiTab