lgsvl/simulator #16: Empty Scenes / Corrupted?

mohamednasrallah, 2018-10-17:
I followed the instructions in the readme and kept getting this problem.
The scenes I select do not load anything in the hierarchy.
I have tried this on multiple clean installs, on my pc, on other pcs.
Am I doing something wrong?

Attached is a screenshot with "SanFrancisco" selected as the scene, and the console with the errors.
Sometimes I get a lot more errors than that, saying that some assets could not be read. Not sure if that was also a problem with serialisation or corruption.

[image]

andytolst, 2018-10-17:
@moenasrallah Do you have GIT LFS installed ? you should see a lot of these while cloning:
```
Downloading Assets/Meshes/BasicCar.fbx (504 KB)
Downloading Assets/Meshes/BasicCar_New.fbx (403 KB)
Downloading Assets/Meshes/BoxTruck_V1_WIP.fbx (1.4 MB)
Downloading Assets/Meshes/Buildings/002/002.fbx (26 KB)
Downloading Assets/Meshes/Buildings/002A/002A.fbx (53 KB)
Downloading Assets/Meshes/Buildings/003/003.fbx (18 KB)
Downloading Assets/Meshes/Buildings/003A/003A.fbx (45 KB)
...
```

moenasrallah, 2018-10-17:
@andytolst 

Thanks for pointing that out. It turns out that I followed the current readme too closely. I had git lfs installed, but the readme said to "git clone" not "git lfs clone"

I cloned it again doing that, then built again, and everything is working. Thanks!

closed 2018-10-17 by moenasrallah