lgsvl/simulator #19: build issue

yangydavid, 2018-10-19:
Hi, 

I met such an issue when trying to build the project following the command below: 
```
mkdir build
 /opt/Unity/Editor/Unity \
    -batchmode \
    -nographics \
    -silent-crashes \
    -quit \
    -buildDestination ./build/simulator \
    -buildTarget Linux64 \
    -executeMethod BuildScript.Build \
    -projectPath . \
    -logFile /dev/stdout
```

The errors shows the following:
```
Aborting batchmode due to failure:
executedMethod class 'BuildScript' could not be found.
Argument was -executedMethod BuildScripts. Build 
```

Could you provide any hints on how to solve the issue? 

Best,
Thank you all!

martins-mozeiko, 2018-10-19:
The issue is that your git did not correctly download files under LFS storag.e You need to correctly install git-lfs. Please follow instructions on https://git-lfs.github.com/  Make sure you run `git lfs install` command. After that clone repository again, and it should work.

saintlrr, 2018-10-26:
@martins-mozeiko  Hi, I have another issue that i can not pass the build process. the following is the error i got: 
```
/apollo/modules/control/common/BUILD:5:1: undeclared inclusion(s) in rule '//modules/control/common:control_gflags':
this rule is missing dependency declarations for the following files included by 'modules/control/common/control_gflags.cc':
  'bazel-out/local-dbg/genfiles/external/com_github_gflags_gflags/gflags/gflags.h'
  'bazel-out/local-dbg/genfiles/external/com_github_gflags_gflags/gflags/gflags_declare.h'
  'bazel-out/local-dbg/genfiles/external/com_github_gflags_gflags/gflags/gflags_google.h'.

```
Could you give me some hints on why this is happening?

martins-mozeiko, 2018-10-26:
This sometimes happens when apollo build cannot resume previous build. The solution is to do clean build. To do so, first run `./apollo.sh clean` and then run build command again.

moenasrallah, 2018-10-27:
@yangydavid 

I came across the same exact issue as you before. 
I took @martins-mozeiko 's suggestion and recloned a fresh copy. Seems to fix the issue. I'm assuming some files during lfs downloads were corrupted/missing and git didnt pick up on them even with git status :confused: 

Let us know if re-cloning and retrying to build worked!

yangydavid, 2018-10-29:
@martins-mozeiko Thanks! I tried the command and it works! I think the issue can be closed.

closed 2018-10-29 by yangydavid