lgsvl/simulator #20: Can not run rosbridge

lesun90, 2018-10-24:
I follow the steps, but could not run the rosbridge script

`[rosbridge_websocket.launch] is neither a launch file in package [rosbridge_server] nor is [rosbridge_server] a launch file name
The traceback for the exception was written to the log file`

martins-mozeiko, 2018-10-24:
Are you trying to run Apollo?
Please make sure you cloned submodule in ros_pkgs folder correctly.

This is done by using `--recursive` argument to git clone.
If you have cloned without this argument, then you can update submodule with following commands:

```
git submodule init
git submodule update
```

lesun90, 2018-10-25:
Solved.
Thank you.

closed 2018-10-25 by lesun90

yangydavid, 2018-10-30:
@martins-mozeiko  Hi, I am having the same issue here and I tried the command you gave above but a new error comes up as follows: 
```
fatal: reference is not a tree: 0f70cefca2d517539b0ffbee8524761649c0220b
Unable to checkout '0f70cefca2d517539b0ffbee8524761649c0220b' in submodule path 'ros_pkgs/src/rosbridge_suite'
```
This error also comes out when I tried to clone the submodule given in the instruction.

Thanks!

martins-mozeiko, 2018-10-30:
It seems we forgot to push update to rosbridge repo. Please try it again, now it is up to date.

yangydavid, 2018-10-30:
@martins-mozeiko  it works now! thanks!