lgsvl/simulator #21: Could not turn on the perception module.

lesun90, 2018-10-25:


hadiTab, 2018-10-25:
What issues did you have with Perception? Any specific errors?
You can try launching perception from a terminal (inside docker) instead of from dreamview to get some print statements. Use the following command:

`./scripts/perception.sh start_fe`

lesun90, 2018-10-26:
My bad, I did not build the cuda for perception.

Thank you

closed 2018-10-26 by lesun90

yangydavid, 2018-10-26:
@lesun90 Hi, I'm kind of having a similar issue, could you elaborate a bit more  about how you built the cuda for perception?

martins-mozeiko, 2018-10-26:
You need to do those things only if you use original Apollo docker image. If you use docker image built by us, you don't need to install Nvidia driver or cuDNN inside docker. Its already there.

lesun90, 2018-10-26:
> You need to do those things only if you use original Apollo docker image. If you use docker image built by us, you don't need to install Nvidia driver or cuDNN inside docker. Its already there.

Yes, exactly, we were using our own image. You guy should put a note in readme so people aware of this. I think anyone who tries your simulator already uses a docker image from apollo before.

yangydavid, 2018-10-26:
@lesun90 @martins-mozeiko Hi, thanks for the comments, while the issue was still not resolved. I am just using the docker-image that you built, so there should be no problem regrading with Nvidia driver or cuDNN. However, the error message I got from `perception.ERROR` is the following: `Check failed: error == cudaSuccess (8 vs. 0)  invalid device function`. By the way, the driver I installed on the machine is 396 and cuda version is 9.2, which is different from the one set in your docker image. I am wondering if there's anything related to it?

martins-mozeiko, 2018-10-26:
If you use our image then what you install on your host does not matter. As long as you have Nvidia driver & cuda installed, everything should work (versions do not matter). For original Apollo image versions matter, and what is in docker should match what is on your host.

What GPU are you using? Maybe it is not supported by Apollo?

lesun90, 2018-10-26:
Maybe your machine has difference nvidia card so it use another version of nvidia, and it will need a difference cuda version

> NVIDIA graphics driver R375 or newer for CUDA 8
> NVIDIA graphics driver R384 or newer for CUDA 9
> NVIDIA graphics driver R390 or newer for CUDA 9.2

I suggest try the following steps:
Install the correct version of your nvida card.
```
sudo add-apt-repository ppa:graphics-drivers
sudo apt-get update
sudo apt-get install nvidia-384
```

Download cuDNN (select correct version for your driver): [](https://developer.nvidia.com/rdp/cudnn-download)
(You will want both the 'Runtime Library' and 'Developer Library')
Install cuDNN
```
sudo dpkg -i libcudnn7_7.1.4.18-1+cuda8.0_amd64.deb
sudo dpkg -i libcudnn7-dev_7.1.4.18-1+cuda8.0_amd64.deb
```
Build Apollo with the GPU option within docker
./apollo.sh build_gpu

In my case it works.

Good luck.

yangydavid, 2018-10-29:
@martins-mozeiko  Hi Martins, yes I am using the nvidia-docker-image. It is still not working. I am wondering if there's anything to do with rosversion? Since I am using Ubuntu 16.04, the only ros I can install is `kinetic`, while the ros inside the docker is `indigo`. I am wondering if this is the cause?

hadiTab, 2018-10-29:
@yangydavid what video card are you using? The version of ros installed on your machine will not affect anything. The docker image will use the indigo installed inside the container.

MartinOrgiazzi, 2019-05-31:
Hi everyone ! I have also a Perception module launch issue and when I launch 
`$ ./scripts/perception.sh start_fe` in another console, I get this : 

`$ ./scripts/perception.sh start_fe
[cyber_launch_6413] INFO Stop cyber launch finished.
[cyber_launch_6413] INFO cyber_launch exit.
cyber_launch stop /apollo/modules/perception/production/launch/perception_all.launch
[cyber_launch_6421] INFO Stop cyber launch finished.
[cyber_launch_6421] INFO cyber_launch exit.
cyber_launch stop /apollo/modules/drivers/tools/image_decompress/launch/image_decompress.launch
[cyber_launch_6425] INFO Stop cyber launch finished.
[cyber_launch_6425] INFO cyber_launch exit.
cyber_launch stop /apollo/modules/transform/launch/static_transform.launch
cyber_launch start /apollo/modules/transform/launch/static_transform.launch &
cyber_launch start /apollo/modules/drivers/tools/image_decompress/launch/image_decompress.launch &
cyber_launch start /apollo/modules/perception/production/launch/perception_all.launch &
morgiazzi@in_dev_docker:/apollo$ [cyber_launch_6429] INFO Launch file [/apollo/modules/drivers/tools/image_decompress/launch/image_decompress.launch]
[cyber_launch_6429] INFO ========================================================================================================================
[cyber_launch_6430] INFO Launch file [/apollo/modules/perception/production/launch/perception_all.launch]
[cyber_launch_6430] INFO ========================================================================================================================
[cyber_launch_6429] INFO Load module [image_decompress] library: [mainboard_default_6429] [CYBER_DEFAULT] conf: [/apollo/modules/drivers/tools/image_decompress/dag/image_decompress.dag] exception_handler: []
[cyber_launch_6430] INFO Load module [perception] library: [perception] [CYBER_DEFAULT] conf: [/apollo/modules/perception/production/dag/dag_streaming_perception.dag] exception_handler: []
[cyber_launch_6429] INFO Start process [mainboard_default_6429] successfully. pid: 6432
[cyber_launch_6429] INFO ------------------------------------------------------------------------------------------------------------------------
[cyber_launch_6428] INFO Launch file [/apollo/modules/transform/launch/static_transform.launch]
[cyber_launch_6428] INFO ========================================================================================================================
[cyber_launch_6428] INFO Load module [static_transform] library: [mainboard_default_6428] [CYBER_DEFAULT] conf: [/apollo/modules/transform/dag/static_transform.dag] exception_handler: []
[cyber_launch_6430] INFO Start process [perception] successfully. pid: 6431
[cyber_launch_6430] INFO ------------------------------------------------------------------------------------------------------------------------
[cyber_launch_6430] INFO Load module [perception_camera] library: [perception] [CYBER_DEFAULT] conf: [/apollo/modules/perception/production/dag/dag_streaming_perception_camera.dag] exception_handler: []
[cyber_launch_6430] INFO Load module [perception_traffic_light] library: [perception_trafficlights] [CYBER_DEFAULT] conf: [/apollo/modules/perception/production/dag/dag_streaming_perception_trafficlights.dag] exception_handler: []
[cyber_launch_6430] INFO Start process [perception_trafficlights] successfully. pid: 6436
[cyber_launch_6430] INFO ------------------------------------------------------------------------------------------------------------------------
[cyber_launch_6428] INFO Start process [mainboard_default_6428] successfully. pid: 6434
[cyber_launch_6428] INFO ------------------------------------------------------------------------------------------------------------------------
[perception: WARNING: Logging before InitGoogleLogging() is written to STDERR] perception: WARNING: Logging before InitGoogleLogging() is written to STDERR
[perception: I0531 16:35:54.378098  6431 module_argument.cc:85] [] command: mainboard -d /apollo/modules/perception/production/dag/dag_streaming_perception.dag /apollo/modules/perception/production/dag/dag_streaming_perception_camera.dag -p perception -s CYBER_DEFAULT ] perception: I0531 16:35:54.378098  6431 module_argument.cc:85] [] command: mainboard -d /apollo/modules/perception/production/dag/dag_streaming_perception.dag /apollo/modules/perception/production/dag/dag_streaming_perception_camera.dag -p perception -s CYBER_DEFAULT 
[perception: I0531 16:35:54.378859  6431 global_data.cc:150] [] host ip: 10.240.9.52] perception: I0531 16:35:54.378859  6431 global_data.cc:150] [] host ip: 10.240.9.52
[perception: I0531 16:35:54.379588  6431 module_argument.cc:61] [] binary_name_ is mainboard, process_group_ is perception, has 2 dag conf] perception: I0531 16:35:54.379588  6431 module_argument.cc:61] [] binary_name_ is mainboard, process_group_ is perception, has 2 dag conf
[perception: I0531 16:35:54.379598  6431 module_argument.cc:64] [] dag_conf: /apollo/modules/perception/production/dag/dag_streaming_perception.dag] perception: I0531 16:35:54.379598  6431 module_argument.cc:64] [] dag_conf: /apollo/modules/perception/production/dag/dag_streaming_perception.dag
[perception: I0531 16:35:54.379601  6431 module_argument.cc:64] [] dag_conf: /apollo/modules/perception/production/dag/dag_streaming_perception_camera.dag] perception: I0531 16:35:54.379601  6431 module_argument.cc:64] [] dag_conf: /apollo/modules/perception/production/dag/dag_streaming_perception_camera.dag
[... cut ...]
[cyber_launch_6430] ERROR Process [perception] has finished. [pid 6431, cmd mainboard -d /apollo/modules/perception/production/dag/dag_streaming_perception.dag /apollo/modules/perception/production/dag/dag_streaming_perception_camera.dag -p perception -s CYBER_DEFAULT].
[cyber_launch_6430] ERROR Process [perception_trafficlights] has finished. [pid 6436, cmd mainboard -d /apollo/modules/perception/production/dag/dag_streaming_perception_trafficlights.dag -p perception_trafficlights -s CYBER_DEFAULT].
[cyber_launch_6430] INFO All processes has died.
[cyber_launch_6430] INFO Cyber exit.
[cyber_launch_6430] INFO All processes have been stopped.
`

Have you a idea of what I can do ?
Thanks

martins-mozeiko, 2019-05-31:
```
Check failed: error == cudaSuccess (2 vs. 0) out of memory
```

This seems to be the real error message. It seems GPU is running out of memory? What kind of GPU are you using?

MartinOrgiazzi, 2019-06-03:
I use a NVIDIA  Quadro M2200 .

MartinOrgiazzi, 2019-06-03:
As the CUDA versions on the docker and the local were different I downgraded the local one to have 8.0 on both. But still it doesn't work. Also mu distribution is Fedora 29, it could be important to notice that, couldn't it ?

martins-mozeiko, 2019-06-03:
Cuda SDK versions in docker and on host does not matter. Apollo docker image only requires your host Nvidia driver (not SDK) to support cuda, and as image is using nvidia-runtime - it will use exactly same version inside docker image.

I think your issue may be that your GPU is not supported. Officially Apollo supports only GTX 1080 (although it runs with 1070/1060 and similar hardware).  According to this table: https://en.wikipedia.org/wiki/CUDA#GPUs_supported the 1080 is Pascal architecture with with 6.1 compute capability. M2200 seems to be Maxwell architecture with 5.2 compute capability. You need to make sure Apollo perception code is compiled with 5.2 support. I suggest clarifying this with Apollo developers.

MartinOrgiazzi, 2019-06-04:
All right I will ask, thank you for the advices. But I think it is a bit strange a could run Apollo with the rosbag of data they attached and not in live with the simulator isn't it ?

martins-mozeiko, 2019-06-04:
Are you running simulator in same laptop as Apollo? If Apollo works without simulator then the issue probably is not enough memory as cuda is already saying. Both Simulator and Apollo wants to use GPU at the same time.

leFeuilluZer, 2019-06-05:
Yes I am. 
I guess I have no choice but moving on other GPU if I want to run everything well then...

Thanks for the help @martins-mozeiko

leFeuilluZer, 2019-06-05:
However, even if I launch only one component in the perception module, I don't have a GPU out of memory error message but it still crashes  : 
[image]

martins-mozeiko, 2019-06-05:
I can suggest running perception process under gdb, and when it crashes check the callstack with `bt` command. Maybe that will give hint what is crashing here.

leFeuilluZer, 2019-06-06:
[image]