lgsvl/simulator #22: import apollo map into simulator

lesun90, 2018-10-25:
Hi,

We have a custom Apollo map, and we want to use that map for the simulator. We don't really need anything fancy like the building or something like that, just the road is enough, the purpose is to test our motion planning. Do you guy plan to have some feature to convert Apollo map into the simulator, or if not can you provide some instruction of how to construct the map for the simulator.

Thank you.

martins-mozeiko, 2018-10-26:
In current implementation simulator does not import maps from Apollo. The workflow is actually opposite - we create map in simulator, annotate it (lanes, center-lines, traffic lights, etc) and then export map information to Apollo format.

Importing basic road structure from Apollo format seems to be useful feature, we could consider implementing something like that in future, but I cannot give any estimates on it.

Currently your only option is to create a new scene in Unity Editor. This includes creating road assets as 3d models, and putting them into scene to create road structure you want. You can check how roads look in SanFrancisco scene - open it in editor and click on any road to see how it is put into scene.

daohu527, 2019-07-26:
Can refer to #199

closed 2020-04-07 by EricBoiseLGSVL