lgsvl/simulator #23: Repo Mirroring Issue

moenasrallah, 2018-10-27:
There seems to be an issue when trying to fork, mirror, or upload your repository on another account (specifically Gitlab). The issue comes from you guys having GIT LFS. 
LFS works nicely with your repository, but does not transfer well over anyone that takes your work and wants to amend it.

Refer to the screenshots attached, and the links to others who have the same kind of issues with LFS

[image]

Also noticed this after doing a git lfs pull... the username area is chopped up

[image]

https://github.com/git-lfs/git-lfs/issues/1935
https://github.com/git-lfs/git-lfs/issues/911

I know this a unsolved LFS error based from Git's side, but I wanted to give some insight and ask if there might be a solution, workaround, or other methods...

Thank you!

martins-mozeiko, 2018-10-27:
git lfs mirroring should work correctly between gitlab and github. We use internal gitlab for our primary repository. And we do sync everything to github from time to time.

I believe the error you see is because git lfs by default does not download lfs objects that are not reachable from current commit (for example if lfs file a has been changed multiple times). It downloads only files that are reachable from currently checked out commit. This saves a lot of time because usually you want to just fetch latest source, not whole history of repo, especially for large files like Unity scene or textures, or meshes, etc.

To download all lfs objects run this:
```
git lfs fetch --all github_remote
```

Then to upload all lfs objects (not only for those that are currently reachable), run this:
```
git lfs push --all gitlab_remote
```

This will create 1:1 mirror repository.

If something is still missing, please paste the lfs hash here as text, so we can check if we accidentally forgot to push something to github from our repo.

moenasrallah, 2018-10-27:
That did the trick!  :tada: 
Glad that you have the answers to lfs issues :smile: 

Support online for lfs problems almost never have any good solutions. :frowning_face: 
This was easy and upfront, and I'm sure others will find it useful.

Thank you Martins!

closed 2018-10-27 by moenasrallah