lgsvl/simulator #25: velodyne point is strange.

hakuturu583, 2018-11-01:
[image]

I tested your simulator and visualize pointcloud from Velodyne Simulation.
I think this pointcloud height is too low.

hakuturu583, 2018-11-01:
Please watchi this video.
https://www.youtube.com/watch?v=FvqGzCRTpSE

hakuturu583, 2018-11-02:
I tested your simulator with Autoware and I found that the pointcloud is height is too low, so Autoware's localization algorithm fails.

hadiTab, 2019-01-04:
I'm closing this issue as it is related to the old LiDAR implementation which has been replaced. To avoid confusion, please open a new issue if similar problems occur with the current implementation of the LiDAR. Thanks.

closed 2019-01-04 by hadiTab