lgsvl/simulator #26: Planning & Control

yangydavid, 2018-11-02:
Dear developers: 

I am interested in the algorithm of planning and control you guys implemented in your Apollo's repo. Could you share some literature on the algorithm being implemented? 

Thanks so much!

yangydavid, 2018-11-02:
@hadiTab Cool, but I found that you have a different planners here as "on road" and "open space", what the difference between them and what is their relationship with EM or Lattice planner?

hadiTab, 2018-11-02:
The "open space" planner was recently added by Apollo, I have not used it but they do cite a paper on it (here)[https://github.com/lgsvl/apollo/blob/simulator/modules/planning/planner/open_space/open_space_planner.h#L37].

The "on road" planner is the EM Planner. For some reason, in the latest apollo updates they changed the name of EM Planner in all of the config files, but if you track down the creation of planner objects in the planner factory, the EM Planner is created whenever the config file calls for the "on road" planner.

These were both developed by Apollo, not us.

yangydavid, 2018-11-06:
@hadiTab  Thanks!

closed 2018-11-06 by yangydavid