lgsvl/simulator #27: build error native installation Ubuntu 16.04

ajithkrishnan, 2018-11-07:
Hi,

I am trying to install the simulator natively on my system (Ubuntu 16.04 LTS). I followed the instructions given in build-instructions.md. 

Specifically, i did download and activate the Unity Engine using my credentials; i cloned the simulator repo after i initialized git lfs (using git lfs install); and after changing to the cloned directory, i used this command to build it: 
```sh
 mkdir build
 /opt/Unity/Editor/Unity2 \
    -batchmode \
    -nographics \
    -silent-crashes \
    -quit \
    -buildDestination ./build/simulator \
    -buildTarget Linux64 \
    -executeMethod BuildScript.Build \
    -projectPath . \
    -logFile /dev/stdout
```
But I keep getting the following compiler errors:

```sh
-----CompilerOutput:-stdout--exitcode: 1--compilationhadfailure: True--outfile: Temp/Assembly-CSharp.dll
Compilation failed: 1 error(s), 1 warnings
-----CompilerOutput:-stderr----------
Assets/Scripts/VehicleController.cs(56,19): warning CS0108: `VehicleController.name' hides inherited member `UnityEngine.Object.name'. Use the new keyword if hiding was intended
/opt/Unity2/Editor/Data/Managed/UnityEngine/UnityEngine.CoreModule.dll (Location of the symbol related to previous warning)
Assets/Scripts/UserInterfaceSetup.cs(147,44): error CS0103: The name `Demo' does not exist in the current context
-----EndCompilerOutput---------------
- Finished compile Library/ScriptAssemblies/Assembly-CSharp.dll
Assets/Scripts/VehicleController.cs(56,19): warning CS0108: `VehicleController.name' hides inherited member `UnityEngine.Object.name'. Use the new keyword if hiding was intended
 
(Filename: Assets/Scripts/VehicleController.cs Line: 56)

Assets/Scripts/UserInterfaceSetup.cs(147,44): error CS0103: The name `Demo' does not exist in the current context
 
(Filename: Assets/Scripts/UserInterfaceSetup.cs Line: 147)

executeMethod class 'BuildScript' could not be found.
Argument was -executeMethod BuildScript.Build
 
(Filename: /home/builduser/buildslave/unity/build/Runtime/Utilities/Argv.cpp Line: 295)

Aborting batchmode due to failure:
executeMethod class 'BuildScript' could not be found.
Argument was -executeMethod BuildScript.Build
```

I have already tried the solutions suggested here : https://github.com/lgsvl/simulator/issues/13
But to no avail. 

According to the error it cannot find the class BuildScript defined in the file ~/simulator/Assets/Scripts/Editor/BuildScript.cs, although i did specify the projectPath as ~/simulator.

I also tried the following:
1. Providing a full path as the project path instead of relative path
2. Cleared the Temp directory and other associated files and rebuilding the repo from scratch
3. using ```sudo -E -u \#${HOST_UID} -g \#${HOST_GID} /opt/Unity2/Editor/Unity``` instead (found this in unity_activate.sh in the Docker folder)
4. Building the repo from by opening it in the Unity Editor as per the main README.md. the project builds in this case but i still get this error when i try to enter "play mode" -
```sh
Assets/Scripts/UserInterfaceSetup.cs(147,44): error CS0103: The name `Demo' does not exist in the current context
```

None of these seem to work. Please advice.

Ajith

martins-mozeiko, 2018-11-07:
The actual error is in UserInterfaceSetup.cs about that "Demo" namespace. Not BuildSciript.

It seems we have pushed to github something that does not compile. Please delete line 147 in `Assets/Scripts/UserInterfaceSetup.cs` file and try again. We'll try to push fixed code as soon as possible.

ajithkrishnan, 2018-11-09:
@martins-mozeiko Thanks. The error was resolved after the new commit.

hadiTab, 2018-11-09:
I'm closing this issue as it seems to have been resolved.

closed 2018-11-09 by hadiTab