lgsvl/simulator #28: How to change the parameters of the vehicle

Chizhang1, 2018-11-08:
Hi, 
Thanks for your great job. I want to config the vehicle to match the MKZ example in Apollo. In #14 it is mentioned that the configuration of vehicle can be modified in Prefabs/RobotCandidates/XE_Rigged-apollo.prefab, but I did not figure it out. Could you help me figure it out? Thanks in advance.

hadiTab, 2018-11-08:
@Chizhang1 what exactly do you mean by vehicle configuration? In [#14](https://github.com/lgsvl/simulator/issues/14) @martins-mozeiko explains how to change sensor configurations. Are you looking to modify the sensor configurations to match the MKZ?
The vehicle in the simulator is a model based on a Jaguar XE and you won't be able to easily modify the model to match an MKZ in terms of dimensions, wheel separation, etc.

At the moment we do not have an MKZ model that you would be able to simply switch to. If you need an MKZ the best approach would probably be to find a model online (probably would need to purchase it) and import it into Unity.

Chizhang1, 2018-11-08:
@hadiTab I am looking for a MKZ model in the simulator. Thanks for your response.

martins-mozeiko, 2019-08-19:
With new simulator [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) release you are able to modify vehicle sensor configuration in WebUI. See documentation [here](https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-change-the-configuration-of-a-vehicle) and [here](https://www.lgsvlsimulator.com/docs/sensor-json-options/). If you have any more questions please open new issue.

closed 2019-08-19 by martins-mozeiko