lgsvl/simulator #29: Invisible Walls

moenasrallah, 2018-11-09:
Is there a purpose to the car not driving in certain areas of the map but NPCs can? I would drive forward when I start the simulation, and eventually an invisible wall will stop my ego vehicle. Other cars drive through that wall but I can't get over it (seems like a map restriction).

Question is... is this a bug or on purpose? If on purpose, how to disable it properly without breaking anything that might depend on it if it has a specific use

martins-mozeiko, 2018-11-09:
It is on purpose. These walls are there to limit car to area where we have annotated the vector map. Currently it does not cover all the available map, but we are planning to do that for whole map.

You can see this map in our Apollo data repository: https://github.com/lgsvl/apollo-simulator-map/blob/master/background.jpg
[image]

To driver further on the map currently you have two choices:

1) Drive around road blocks on sidewalk.

2) Open scene in Unity editor and disable or delete roadblocks. It will not break anything.

In this case only manual driving will work, because Apollo and Autoware needs annotated vector map for AD.

andytolst, 2018-11-09:
actually it's even smaller (see orange lines):
[image]

moenasrallah, 2018-11-09:
@martins-mozeiko thank you. That helps
@andytolst thank you for the additional clarification