lgsvl/simulator #30: Build Issue

judyqiu1996, 2018-11-09:
Hi, every time when I try to build the simulator, it will have the error message following:

build command
```
 /opt/Unity/Editor/Unity \

>     -batchmode \

>     -nographics \

>     -silent-crashes \

>     -quit \

>     -buildDestination ./build/simulator \

>     -buildTarget Linux64 \

>     -executeMethod BuildScript.Build \

>     -projectPath . \

>     -logFile /dev/stdout
```

The output of command
```
DisplayProgressbar: Unity license

[1108/014945:ERROR:browser_main_loop.cc(161)] Running without the SUID sandbox! See https://code.google.com/p/chromium/wiki/LinuxSUIDSandboxDevelopment for more information on developing with the sandbox on.

Cancelling DisplayDialog: Failed to activate/update license. Missing or bad username and password. Please try again using valid credentials or contact support@unity3d.com

This should not be called in batch mode.

(Filename: /home/builduser/buildslave/unity/build/Editor/Platform/Linux/EditorUtility.cpp Line: 241)
```

I follow the instructions in https://github.com/lgsvl/simulator/blob/master/Docs/build-instructions.md.  But still face this issue for a very long time.

martins-mozeiko, 2018-11-09:
You need to open Unity at least once to activate it. It will require logging in with your Unity account or creating new one.

Run `/opt/Unity/Editor/Unity` and follow instructions on screen.

If you have bought Unity license (instead of using free one) then you should be able to pass your license information with extra arguments:

    -serial SB-XXXX-XXXX-XXXX-XXXX-XXXX \
    -username "JoeBloggs@example.com" \
    -password "MyPassw0rd" \

See https://docs.unity3d.com/Manual/CommandLineArguments.html for more information.

martins-mozeiko, 2018-11-09:
This message is shown when other errors already happened before. Please post here few lines above - it should have text `Error` somewhere.

judyqiu1996, 2018-11-09:
@martins-mozeiko  Here is the error message.

```
-----CompilerOutput:-stdout--exitcode: 1--compilationhadfailure: True--outfile: Temp/Assembly-CSharp.dll
Compilation failed: 8 error(s), 1 warnings
-----CompilerOutput:-stderr----------
Assets/Scripts/ApolloAutoMode.cs(2,7): error CS0246: The type or namespace name `WebSocketSharp' could not be found. Are you missing an assembly reference?
Assets/Scripts/Ros/Bridge.cs(12,7): error CS0246: The type or namespace name `WebSocketSharp' could not be found. Are you missing an assembly reference?
Assets/UnityPackages/JsonNetSample/JsonNetSample.cs(5,7): error CS0246: The type or namespace name `Newtonsoft' could not be found. Are you missing an assembly reference?
Assets/UnityPackages/JsonNetSample/JsonNetSample.cs(6,7): error CS0246: The type or namespace name `Newtonsoft' could not be found. Are you missing an assembly reference?
Assets/Scripts/Ros/Bridge.cs(38,9): error CS0246: The type or namespace name `WebSocket' could not be found. Are you missing `System.Net.WebSockets' using directive?
Assets/Scripts/Ros/Bridge.cs(113,37): error CS0246: The type or namespace name `CloseEventArgs' could not be found. Are you missing an assembly reference?
Assets/Scripts/Ros/Bridge.cs(127,37): error CS0246: The type or namespace name `ErrorEventArgs' could not be found. Are you missing `System.IO' using directive?
Assets/Scripts/Ros/Bridge.cs(146,39): error CS0246: The type or namespace name `MessageEventArgs' could not be found. Are you missing `UnityEngine.Networking.PlayerConnection' using directive?
Assets/Scripts/VehicleController.cs(56,19): warning CS0108: `VehicleController.name' hides inherited member `UnityEngine.Object.name'. Use the new keyword if hiding was intended
/opt/Unity/Editor/Data/Managed/UnityEngine/UnityEngine.CoreModule.dll (Location of the symbol related to previous warning)
-----EndCompilerOutput---------------
- Finished compile Library/ScriptAssemblies/Assembly-CSharp.dll
- Finished compile Library/ScriptAssemblies/Assembly-CSharp-Editor-firstpass.dll
Assets/Scripts/ApolloAutoMode.cs(2,7): error CS0246: The type or namespace name `WebSocketSharp' could not be found. Are you missing an assembly reference?
 
(Filename: Assets/Scripts/ApolloAutoMode.cs Line: 2)

Assets/Scripts/Ros/Bridge.cs(12,7): error CS0246: The type or namespace name `WebSocketSharp' could not be found. Are you missing an assembly reference?
 
(Filename: Assets/Scripts/Ros/Bridge.cs Line: 12)

Assets/UnityPackages/JsonNetSample/JsonNetSample.cs(5,7): error CS0246: The type or namespace name `Newtonsoft' could not be found. Are you missing an assembly reference?
 
(Filename: Assets/UnityPackages/JsonNetSample/JsonNetSample.cs Line: 5)

Assets/UnityPackages/JsonNetSample/JsonNetSample.cs(6,7): error CS0246: The type or namespace name `Newtonsoft' could not be found. Are you missing an assembly reference?
 
(Filename: Assets/UnityPackages/JsonNetSample/JsonNetSample.cs Line: 6)

Assets/Scripts/Ros/Bridge.cs(38,9): error CS0246: The type or namespace name `WebSocket' could not be found. Are you missing `System.Net.WebSockets' using directive?
 
(Filename: Assets/Scripts/Ros/Bridge.cs Line: 38)

Assets/Scripts/Ros/Bridge.cs(113,37): error CS0246: The type or namespace name `CloseEventArgs' could not be found. Are you missing an assembly reference?
 
(Filename: Assets/Scripts/Ros/Bridge.cs Line: 113)

Assets/Scripts/Ros/Bridge.cs(127,37): error CS0246: The type or namespace name `ErrorEventArgs' could not be found. Are you missing `System.IO' using directive?
 
(Filename: Assets/Scripts/Ros/Bridge.cs Line: 127)

Assets/Scripts/Ros/Bridge.cs(146,39): error CS0246: The type or namespace name `MessageEventArgs' could not be found. Are you missing `UnityEngine.Networking.PlayerConnection' using directive?
 
(Filename: Assets/Scripts/Ros/Bridge.cs Line: 146)

Assets/Scripts/VehicleController.cs(56,19): warning CS0108: `VehicleController.name' hides inherited member `UnityEngine.Object.name'. Use the new keyword if hiding was intended
 
(Filename: Assets/Scripts/VehicleController.cs Line: 56)

executeMethod class 'BuildScript' could not be found.
Argument was -executeMethod BuildScript.Build
 
(Filename: /home/builduser/buildslave/unity/build/Runtime/Utilities/Argv.cpp Line: 295)

Aborting batchmode due to failure:
executeMethod class 'BuildScript' could not be found.
Argument was -executeMethod BuildScript.Build
```

martins-mozeiko, 2018-11-09:
This log suggests that some binary files were not downloaded from git.

It can happen if [Git LFS](https://git-lfs.github.com) was not installed correctly. Please install it, including running `git lfs install` command so it can install git hooks for your user.

Then to fix repository, you can delete current one & clone repository again. Or you can run `git lfs fetch && git lfs checkout` command on existing folder to download LFS files from github. In this case you will need to do it only once. Next time regular `git pull` will work correctly.

judyqiu1996, 2018-11-14:
@martins-mozeiko Thanks! It solved it. And I am facing another error when I try to run the simulator. 
I run command after I finished the build inside of the simulator folder.
```
roslaunch rosbridge_server rosbridge_websocket.launch
```
And the error message is 
```
[rosbridge_websocket.launch] is neither a launch file in package [rosbridge_server] nor is [rosbridge_server] a launch file name
The traceback for the exception was written to the log file
```
I saw other people is encountering this error and solved it by using `git submodule init && git submodule update`. I tried that but it didn't work. And I tried to run simulator with apollo, when I go into the docker, I could not find ros_pkgs directory. And if I do `catkin_make` outside of the container, it will have error message like this
```
####
#### Running command: "make cmake_check_build_system" in "/home/yujiaq3/test/apollo/ros_pkgs/build"
####
CMake Error: The source directory "/apollo/ros_pkgs/src" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Makefile:498: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

martins-mozeiko, 2018-11-14:
It seems that you don't have `rosbridge_server` package installed on your host machine, that's why it does not run. You need to install it first. Note that we have modified rosbridge_server package for Apollo to work. You can get it in here: https://github.com/lgsvl/rosbridge_suite/

Are you running Apollo from our fork at https://github.com/lgsvl/apollo or are you trying to use it with original Apollo repository? Please follow instructions in our Apollo readme if you want to run it with our simulator. It contains whole environment needed for simulator - including `ros_pkgs` directory and our modified rosbridge_server, and few other tweaks in Apollo.

judyqiu1996, 2018-11-14:
@martins-mozeiko Do you know to install `rosbridge_server` on my machine? Because I didn't find any instructions on https://github.com/lgsvl/rosbridge_suite/. 

And I am running Apollo following your instructions at https://github.com/lgsvl/apollo. But after going into the container by using `./docker/scripts/dev_into.sh`, then I couldn't find `ros_pkgs` folder.

martins-mozeiko, 2018-11-14:
Installing rosbridge_server on your machine will work only if you have Apollo also installed on your host, without docker. It is better to use docker environment for this because Apollo depends on specific versions of packages, so it is much easier.

It is very strange that you don't have ros_pkgs folder. Are you sure you have cloned Apollo from https://github.com/lgsvl/apollo and not upstream? You can see ros_pkgs folder right here: https://github.com/lgsvl/apollo/tree/simulator/ros_pkgs

Could you please run these three commands from apollo folder and show the output here?

```
git remote show origin -n
git branch
ls -l
```

judyqiu1996, 2018-11-14:
@martins-mozeiko The output of three commands is 
```
yujiaq3@mankndn2-HP-Z220-CMT-Workstation:~/test/apollo$ git remote show origin -n
* remote origin
  Fetch URL: https://github.com/lgsvl/apollo.git
  Push  URL: https://github.com/lgsvl/apollo.git
  HEAD branch: (not queried)
  Remote branches: (status not queried)
    20180730_1511_dev
    apollo
    auto
    cidi_dev
    cidi_dev_stable
    demo_dev
    master
    mobileye_radar
    pred_lab_dnn
    r1.0.0
    r1.5.0
    r2.0.0
    r2.5.0
    r3.0.0
    simulator
    v3.1_dev
  Local branch configured for 'git pull':
    simulator merges with remote simulator
  Local ref configured for 'git push' (status not queried):
    (matching) pushes to (matching)
yujiaq3@mankndn2-HP-Z220-CMT-Workstation:~/test/apollo$ git branch
* simulator
yujiaq3@mankndn2-HP-Z220-CMT-Workstation:~/test/apollo$ ls -l
total 268
-rwxrwxr-x  1 yujiaq3 yujiaq3   4623 Nov  1 23:39 apollo_docker.sh
-rw-rw-r--  1 yujiaq3 yujiaq3 100486 Nov  1 23:39 apollo.doxygen
-rw-rw-r--  1 yujiaq3 yujiaq3   9919 Nov  1 23:39 Apollo_README.md
-rwxrwxr-x  1 yujiaq3 yujiaq3  24724 Nov  1 23:39 apollo.sh
-rw-rw-r--  1 yujiaq3 yujiaq3    934 Nov  1 23:39 AUTHORS.md
lrwxrwxrwx  1 yujiaq3 yujiaq3     90 Nov  1 23:44 bazel-apollo -> /home/yujiaq3/.cache/bazel/_bazel_yujiaq3/540135163923dd7d5820f3ee4b306b32/execroot/apollo
lrwxrwxrwx  1 yujiaq3 yujiaq3    114 Nov  1 23:44 bazel-bin -> /home/yujiaq3/.cache/bazel/_bazel_yujiaq3/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/bin
lrwxrwxrwx  1 yujiaq3 yujiaq3    119 Nov  1 23:44 bazel-genfiles -> /home/yujiaq3/.cache/bazel/_bazel_yujiaq3/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/genfiles
lrwxrwxrwx  1 yujiaq3 yujiaq3    100 Nov  1 23:44 bazel-out -> /home/yujiaq3/.cache/bazel/_bazel_yujiaq3/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out
lrwxrwxrwx  1 yujiaq3 yujiaq3    119 Nov  1 23:44 bazel-testlogs -> /home/yujiaq3/.cache/bazel/_bazel_yujiaq3/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs
-rw-rw-r--  1 yujiaq3 yujiaq3    100 Nov  1 23:39 BUILD
-rw-rw-r--  1 yujiaq3 yujiaq3   3867 Nov  1 23:39 CONTRIBUTING.md
-rw-rw-r--  1 yujiaq3 yujiaq3    309 Nov  1 23:39 CPPLINT.cfg
drwxrwxr-x  6 yujiaq3 yujiaq3   4096 Nov  2 02:07 data
drwxrwxr-x  6 yujiaq3 yujiaq3   4096 Nov  1 23:39 docker
drwxrwxr-x  8 yujiaq3 yujiaq3   4096 Nov  1 23:39 docs
-rw-rw-r--  1 yujiaq3 yujiaq3  11425 Nov  1 23:39 LICENSE
drwxrwxr-x 21 yujiaq3 yujiaq3   4096 Nov  1 23:39 modules
drwxr-xr-x  3 yujiaq3 yujiaq3   4096 Nov  2 02:07 py_proto
-rw-rw-r--  1 yujiaq3 yujiaq3   3322 Nov  1 23:39 README_cn.md
-rw-rw-r--  1 yujiaq3 yujiaq3   6450 Nov  1 23:39 README.md
-rw-rw-r--  1 yujiaq3 yujiaq3   5734 Nov  1 23:39 RELEASE.md
drwxrwxr-x  5 yujiaq3 yujiaq3   4096 Nov  2 02:08 ros_pkgs
drwxrwxr-x  2 yujiaq3 yujiaq3   4096 Nov  1 23:39 scripts
drwxrwxr-x  4 yujiaq3 yujiaq3   4096 Nov  1 23:39 third_party
drwxrwxr-x  4 yujiaq3 yujiaq3   4096 Nov  1 23:39 tools
-rw-r--r--  1 yujiaq3 yujiaq3   5648 Nov 13 18:28 WORKSPACE
-rw-rw-r--  1 yujiaq3 yujiaq3   5656 Nov  1 23:39 WORKSPACE.in
```
I can find `ros_pkgs` folder outside of docker, like this 
```
apollo_docker.sh  Apollo_README.md  AUTHORS.md	  bazel-bin	  bazel-out	  BUILD		   CPPLINT.cfg	docker	LICENSE  py_proto      README.md   ros_pkgs  third_party  WORKSPACE
apollo.doxygen	  apollo.sh	    bazel-apollo  bazel-genfiles  bazel-testlogs  CONTRIBUTING.md  data		docs	modules  README_cn.md  RELEASE.md  scripts   tools	  WORKSPACE.in
```
But I will face this error
```
####
#### Running command: "make cmake_check_build_system" in "/home/yujiaq3/test/apollo/ros_pkgs/build"
####
CMake Error: The source directory "/apollo/ros_pkgs/src" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Makefile:498: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```
 But once I go into the docker, then I can only see 
```
AUTHORS.md       LICENSE       WORKSPACE       apollo_docker.sh  bazel-out            callgrind.out.28691  callgrind.out.7617  control_prof.pdf  docker                     help?name=sensor%20data.2  prediction_prof.pdf  scripts
BUILD            README.md     WORKSPACE.in    bazel-apollo      bazel-testlogs       callgrind.out.28822  callgrind.out.7798  data              docs                       help?name=sensor%20data.3  py_proto             third_party
CONTRIBUTING.md  README_cn.md  apollo.doxygen  bazel-bin         callgrind.out.28450  callgrind.out.4512   callgrind.out.7891  demo_2.5.bag      help?name=sensor%20data    modules                    ros_statistics_msgs  tools
CPPLINT.cfg      RELEASE.md    apollo.sh       bazel-genfiles    callgrind.out.28674  callgrind.out.4558   callgrind.out.7905  demo_2.5.np.bag   help?name=sensor%20data.1  planning_prof.pdf          rosprofiler
```
And you can ignore the `callgrind.out.*` files, but the `ros_pkgs` folder will disappear.

Actually, I just don't know how to install `rosbridge_server` because there are no instructions in the  https://github.com/lgsvl/rosbridge_suite/. And you said `rosbridge_server` is needed to run the rosbridge.

martins-mozeiko, 2018-11-14:
We are not really supporting case when bridge runs outside of container. It could work when you modify its build scripts, but it is a bit tricky because it will depend on apollo files that is build inside docker.

As for error - there seems to be an issue how you are running docker. Maybe you are running different docker image than we are providing? Files inside docker should be exactly same as outside. If you are using custom apollo docker image, then you will need to update it manually to have ros_pkgs folder with rosbridge_server inside.

Are you launching it by running `./docker/scripts/dev_start.sh` from same folder where lgsvl/apollo repository is cloned (should be /home/yujiaq3/test/apollo/ folder)?

Can you please show output of `docker ps -a` command while container is running? It should show that `apollo_dev` container is created from `lgsvl/apollo:latest` image.

And `docker images | grep lgsvl` should show that `lgsvl/apollo:latest` has `10d0ae2b33f9` image id.

judyqiu1996, 2018-11-14:
@martins-mozeiko Thanks, I think I can run ` ./scripts/rosbridge.sh`. The output would be 
```
yujiaq3@in_dev_docker:/apollo$ ./scripts/rosbridge.sh
... logging to /home/yujiaq3/.ros/log/c3137f20-e7bc-11e8-93bc-d4c9efef2941/roslaunch-in_dev_docker-16274.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://in_dev_docker:33439/

SUMMARY
========

PARAMETERS
 * /rosapi/params_glob: [*]
 * /rosapi/services_glob: [*]
 * /rosapi/topics_glob: [*]
 * /rosbridge_websocket/address: 
 * /rosbridge_websocket/authenticate: False
 * /rosbridge_websocket/bson_only_mode: False
 * /rosbridge_websocket/delay_between_messages: 0
 * /rosbridge_websocket/fragment_timeout: 600
 * /rosbridge_websocket/max_message_size: 134217728
 * /rosbridge_websocket/params_glob: [*]
 * /rosbridge_websocket/port: 9090
 * /rosbridge_websocket/retry_startup_delay: 5
 * /rosbridge_websocket/services_glob: [*]
 * /rosbridge_websocket/topics_glob: [*]
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    rosapi (rosapi/rosapi_node)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rosbridge_websocket-1]: started with pid [16321]
process[rosapi-2]: started with pid [16363]
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
registered capabilities (classes):
 - rosbridge_library.capabilities.call_service.CallService
 - rosbridge_library.capabilities.advertise.Advertise
 - rosbridge_library.capabilities.publish.Publish
 - rosbridge_library.capabilities.subscribe.Subscribe
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - rosbridge_library.capabilities.advertise_service.AdvertiseService
 - rosbridge_library.capabilities.service_response.ServiceResponse
 - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
/use_sim_time is not set, will not subscribe to simulated time [/clock] topic
/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_server/scripts/rosbridge_websocket:218 <module>() [INFO] [1542165882.195766]: Rosbridge WebSocket server started on port 9090
```
And the next step on github would be `Run the LG SVL Simulator (see instructions in the simulator repository)`. Does that mean running command `  roslaunch rosbridge_server rosbridge_websocket.launch` in simulator folder? I am stilling have `[rosbridge_websocket.launch] is neither a launch file in package [rosbridge_server] nor is [rosbridge_server] a launch file name
The traceback for the exception was written to the log file` this error. Does that mean I need to install `rosbridge_server`?

martins-mozeiko, 2018-11-14:
Excellent! This is expected output from rosbridge.
You don't need to run roslaunch manually, the `./scripts/rosbridge.sh` script is running it (check its contents with `cat`).

Now you can run the simulator and launch Apollo modules. With the help of this rosbridge they will be able to talk to each other.

hadiTab, 2019-01-04:
@judyqiu1996  I'm closing this issue as it seems to have been resolved. Feel free to re-open it if needed. Thanks.

closed 2019-01-04 by hadiTab