lgsvl/simulator #31: Import URDF files

ajithkrishnan, 2018-11-13:
Hi,
I am working on a project where i have to import an existing rover model (complete with meshes, joints and tf tree in the form of urdf files) into the existing simulation. 

Is there an out of the box way to do this? I am relatively new to Unity and would like to avoid having to design the rover from scratch inside the Engine itself.

Also if possible could you provide instructions on how to communicate the data with ROS? eg. I would like to extract the LIDAR data and publish it onto ROS topics.

Thanks.

hadiTab, 2018-11-13:
@ajithkrishnan at the moment there is no simple way to import a rover model over using the URDF. You would unfortunately need to either model the rover or find an existing model elsewhere and import it.
The positions of the sensors can be easily modified to match your URDF, however, things like the distance between the wheels, steering angle, wheel diameter, etc. would need also need to match your URDF if you intend on simulating the vehicle motion accurately.

ajithkrishnan, 2018-11-14:
@hadiTab Thanks for responding. I have found that Unity has a URDF Importer in its App store that runs on Linux. Hopefully that should work for now. 

As a side note, could you give instructions on how to read/write sensor data in Unity? I would like to publish msgs like PointCloud2, Odometry and Twist messages onto ROS topics so as to process them in ROS or visualize in RVIZ.

hadiTab, 2018-11-14:
@ajithkrishnan we have not tested the URDF importer for Unity yet. Let us know how it goes!

The sensors on the vehicle are currently publishing to rostopics. For example, the lidar message is published on a PointCloud2 message type. The topic names and message types we are using are based on requirements by apollo and autoware.

For example, look at the [source code](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/ImuSensor.cs) for the IMU sensor. This only appears in the vehicle specified for Apollo and publishes a message of type pb_msgs/Imu. This message type is defined in the Ros namespace, in Assets/Scripts/Ros/PbMsgs.cs. If you, for example, wish to publish imu messages using the [Sensor_msgs/Imu](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Imu.html), you would need to define this message type in a similar manner and modify ImuSensor.cs to publish this message type.

I suggest you look at components already in the simulator which publish to get a better idea. They will be under Assets/Scripts/. For example, [GpsDevice.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/GpsDevice.cs)  or [CanBus.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/CanBus.cs).

ajithkrishnan, 2018-11-20:
@hadiTab  URDF Importer works out of the box. So i would recommend using it. Just make sure to convert your xacros files (if any) to urdf, since URDF Importer doesn't support xacro files.

Thank you for the info regarding the sensors. Will take a look into it as soon as i am done importing a working car model into the simulator.