lgsvl/simulator #33: No texture in pedestrian

hakuturu583, 2018-11-16:
In your movie, pedestrian models has textures.
But, in my case, pedestrian models have no textures.
[image]
[image]
[image]

hadiTab, 2018-11-16:
@hakuturu583 the pedestrians are still under development and not yet complete. We have released these "skinless" pedestrians as a first step as they may still be useful to some. The pedestrians will be updated in the future with appropriate textures.

hadiTab, 2018-12-20:
@hakuturu583 we added textures to the pedestrians in the latest release.
I'm closing this issue. Feel free to re-open it if you still have issues. Thanks.

closed 2018-12-20 by hadiTab