lgsvl/simulator #34: Traffic Simulation Vehicle Suddenly Stops on the road.

hakuturu583, 2018-11-16:
Some Vehicle in your traffic simulation suddenly stops on the Load.
It seems the wheel is still rolling.

[image]

moenasrallah, 2018-11-16:
This is explained in this closed issue:

https://github.com/lgsvl/simulator/issues/29

hadiTab, 2018-11-19:
@hakuturu583 we are aware of this issue and are looking into it. It should be resolved in the near future.
As for the invisible stops, they are at the boundary of the map as @moenasrallah mentioned.

hadiTab, 2019-01-04:
This issue should be fixed with the latest simulator. Closing the issue. Feel free to re-open if the issue persists. Thanks.

closed 2019-01-04 by hadiTab