lgsvl/simulator #35: How to prepare multi robot.

hakuturu583, 2018-11-16:
I want to simulate multi autonomous driving vehicle in your sim.
How can I do that??

martins-mozeiko, 2018-11-16:
You can do that by adding multiple vehicles / ros connections in main menu. Click `+` button multiple times, enter different bridge addresses and you should see multiple vehicle spawn in the scene and to connect to separate rosbridge servers.

However note that at this moment enabling all the sensors for multiple vehicles will require a lot of processing power. Very high-end CPU will be required in this case. We are still working on performance optimizations, so it will get better in future.

closed 2018-11-23 by hakuturu583

Engietoday, 2019-01-29:
I got 2 cars connected to 2 instances of rosbridge websocket with different namespaces but all the sensor information are still conflicting with each other. For example the 2 cars are publishing camera feed on the same topic. Any help?