lgsvl/simulator #37: How to configure the vehicle?

ajithkrishnan, 2018-11-20:
Hi,
I would like to make changes to the existing vehicle prefab viz. XE_Rigged-apollo. 

I found the mesh of the car in Assets/Meshes/XE_LowPoly.fbx
I have a van.fbx mesh of a van which has similar dimensions as that of the car. However i found that i could not simply replace these files (by renaming van.fbx to LowPoly.fbx). It does not show up in the editor and no error is thrown on the Console. 
I can't find a Mesh Filter component in the Inspector either, when the XE_Rigged-apollo prefab is selected.

Could you provide instructions on how to achieve this or on how to use the existing scripts and settings on another vehicle?

ruudyLiu, 2018-11-20:
In order to make an existing model work in the simulator it is not enough to just rename a fbx file. What you need to do is to open the .fbx file in a 3D modeling software(ex. Maya) and then replace each individual mesh with the one you want from the van.fbx, keep all of the meshes' names, pivot points and transformation information the same and then export to overwrite the existing .fbx file in simulator. Then when you go back to Unity the prefab should have new prefab hooked up with all scripts with thee same serialized value on it.

ajithkrishnan, 2018-11-21:
@ruudyLiu Thanks for responding. I don't think i will approach the problem this way, since my project doesn't require such a detailed mesh. 

As an alternative, i was thinking of creating my own Prefab using the scripts used in the Jaguar XE. I am not overly concerned about precision regarding Physics, Lighting, Effects etc. or with additional functionalities like headlights, canbus etc. The existing configurations should suffice (with minor tweaks). 

Could you tell me which scripts are the most important, if for the moment i would just like to drive around using the Keyboard? Scripts like VehicleController, CarInputController, VehicleInputController, KeyboardInputController are obvious. But what about other scripts (eg. CanBus) which might be a dependency too? 
Essentially i would like to know which files/dependencies to export into a standalone package, if i require only the minimal functionality for the car.

Thanks.

closed 2018-11-30 by ajithkrishnan

yangjiongfeng, 2019-08-02:
@ajithkrishnan I also want to try to control my car model with keyboard.
How do you configure it?