lgsvl/simulator #39: HD Map Visualization

moenasrallah, 2018-11-26:
Is it possible to visualize the HD map lanes? In your ROSCON video, it showed you guys editing the HD map through the Unity objects you have placed on the roads themselves.

See Image for reference:
[image]

Are the blue visual lines/arrows from a gizmo setting in Unity? Or a script that enables them?

Thank you!

martins-mozeiko, 2018-11-26:
You can view/edit vector information by doing following:

1. choose Window -> Map Tool Panel menu item, new window will show up
2. click "Show/Hide Map" - this will make vector information to appear. It is off by default for performance reasons
3. click on circles to select part of vector line - then you'll be able to move it around (as a whole)
4. click on "Display Handles" checkbox in Inspector when you have selected part of vector line - this will allow to move each individual "circle" to adjust curvature.
5. use add point at zero/end to add new circles at beginning/end of vector line. And use other options like remove point, etc...

See following image that illustrates each point:

[image]

frontw, 2018-12-05:
It's really cool, as visualization, but could this handles be used for hd-map creation and editing. For me it works too slow, is it same for others?

martins-mozeiko, 2018-12-05:
That's exactly what these handles are used for - for hd map creating and editing. The current annotations on current map were done with this tool - both for Apollo and Autoware.
As for how fast Unity renders them - that depends on your hardware. Reasonably good Nvidia GPU is recommended. Also running on Windows helps a lot - it works there faster than on Linux.

frontw, 2018-12-06:
> That's exactly what these handles are used for - for hd map creating and editing. The current annotations on current map were done with this tool - both for Apollo and Autoware.
> As for how fast [...]

Thank you for reply! Now I could confirm that Windows version of Unity3d works better too, and it's feasible to edit map.
But could you, please, give manual or some hints, how to create a new map on empty scene, because Map editor has a lot of buttons, but not all is working for new scene. #45