lgsvl/simulator #40: Create second vehicle controlled via the keyboard, instead of Apollo?

mhct, 2018-11-26:
We are looking into how to create a second vehicle, which should be controlled by a second player, 
via the keyboard. 

Can somebody give some advice on the best way to do that?

martins-mozeiko, 2018-11-26:
Simulator already supports controlling multiple cars. First you add multiple vehicles in main menu. Then in the scene you should see multiple vehicles in top left corner. You can click on them - that will switch which vehicle is active. Keyboard will control vehicle that is currently active.

mhct, 2018-11-26:
That is great.

Adding more cars in the initial menu let me add more cars connected to a rosbridge.
Is it possible that only 1 car is controlled by ros (apollo) and the other has no rosbridge 
attached to it?

martins-mozeiko, 2018-11-26:
Yes, that is possible. Simple leave address for one vehicle to something invalid (`localhost:1111`). It will try to connect and will not succeed. And that is OK. To start the map without connected rosbridge use Left Shift + click on run.

closed 2018-11-27 by mhct