lgsvl/simulator #41: how to add new scenes??

hakuturu583, 2018-11-27:
I built new simulation environment and I want to add my environment to the menu.

[image]

How can I add my environment to the menu??

ruudyLiu, 2018-11-27:
1. Go to find and select a scriptableobject type project asset file in the path of "Assets/ScriptableObjects/DefaultAssetBundleSettings.asset"
2. Add one more element and drag your new scene(.unity file) into the Scene Asset field of the new element(another sprite field can be left empty) and save the change
[image]

3. open unity Build Settings dialog, add your new scene there as well. This step is only required if you want to test your scene through editor. When you want to test your scene in standalone this step is not required.
[image]

Let me know if that works

hakuturu583, 2018-11-28:
I tested as you said, but new map was not appeared....
[image]

[image]

ruudyLiu, 2018-11-29:
Did you change the DefaultAssetBundleSettings.asset map list? As I don't see you post any picture showing this step. Both DefaultAssetBundleSettings.asset and the Build Settings need to be changed in order to make this work.

hakuturu583, 2019-01-18:
I succeed. Thanks!!

martins-mozeiko, 2019-01-21:
Can you please confirm that you did following steps Ruudy explained above?

1. Select `Assets/ScriptableObjects/DefaultAssetBundleSettings.asset` in project panel.
2. In inspector panel increment `Size` for `Maps` array.
3. Set `Scene Asset` property for new element to the new `.scene` file you want.
4. Save the project, close Editor and then run the build on command-line with ` -executeMethod BuildScript.Build` argument.

See the following screenshot for how it should look like:

[image]

hakuturu583, 2019-01-22:
I succeed Thanks!!

closed 2019-01-22 by hakuturu583