lgsvl/simulator #42: Connecting additional sensors to rosbridge

ajithkrishnan, 2018-11-30:
Hi,
I have been working on replacing the existing Jaguar XE prefab with my own vehicle model and was successful in creating a prefab and getting it to run in the simulation itself. However i want to extract the sensor data and publish it to ROS topics. I setup up the SingleROSConnection component with necessary values, however i found out that to publish the values i need to also setup the components in RobotSetup.cs in the NeedsBridge list. 
This list is defined as follows: public List<Component> NeedsBridge;

By default the 12 Elements of this list (for the Jaguar XE prefab) are LidarSensor.cs, GPSDevice.cs etc). But when i try to select the corresponding elements for my vehicle, Unity automatically selects the Transform Component and not the corresponding C# script. 

I did not face this problem with any other lists since the datatype of the list automatically selects the required Component (Transform, Collider etc). But i feel since the datatype of NeedsBridge is <Component> it automatically selects the Transform Component.

NB: I face the same problem when i try to make changes to the NeedsBridge list in the Jaguar XE prefab.  

Could someone give me directions on how to fix this?

Thanks

ajithkrishnan, 2018-11-30:
I found a dirty fix for this problem. In RobotSetup.cs, instead of one list for all the sensors, i created a variable for each sensor.
That is i replaced : 
public List<Component> NeedsBridge;
with : 
public LidarSensor LidarNeedsBridge;
public ImuSensor ImuNeedsBridge;
.... and so on

Then i copied the operations performed on NeedsBridge list on each of these variables. That has essentially solved the problem and i can now select the C# scripts in unity instead of the Transform Component.

If there is a cleaner approach, do let me know.

closed 2018-11-30 by ajithkrishnan

martins-mozeiko, 2018-11-30:
If I understand correctly you have trouble dragging specific component into this list?

If so - then you need to open 2 inspector windows. Select the game object that has RobotSetup component and click "Lock" icon in one of inspector window. Then select game object that has component you want to drag into this NeedsBridge list. And then drag the component (not whole game object) from second inspector window into the this list.

Here's an illustration:

[image]

(1) First select XE_Rigged (car) object and click lock in first inspector
(2) Then select new sensor
(3/4) drag sensor script from inspector (not whole game object from scene hierarchy) into slot in the first inspector.

ajithkrishnan, 2018-12-05:
@martins-mozeiko This worked, and is a cleaner approach. Thanks :)