lgsvl/simulator #43: Apollo : Route Editing tab not accessible

ajithkrishnan, 2018-12-05:
Hi,
I am aware that the LG apollo fork is in a different repo, however I did not find an "Issues" tab there, so i am posting my query here.

I followed the instructions in the README.md and was able to bring up the simulation, drive the car around and toggle the sensors. However, I am still facing problems with implementing the Apollo Stack. I selected "LG SVL 1080" and "San Francisco" in the Dreamviewer and toggled on all the modules in the "Module Controller" tab as per the instructions. However, the "Route Editing" tab remains greyed out. I am able to access the "Route Editing" tab occasionally and send waypoint goals, but only if i restart the simulation and the docker container (and even then it doesn't seem to run most of the time). Furthermore, The modules in the "Module Controller" tab automatically toggle off after i select them.

martins-mozeiko, 2018-12-05:
Route Editing should be enabled when car can localize itself on the map. Have you turned on GPS and IMU sensors in simulator.? Then turn on Localization module. The car should now show on the map.

Note that changing car and map choices will turn off all the modules. That is what dreamview is doing in Apollo automatically.

If module automatically toggles off that means the process is either exiting or crashing. You can check the log files. For example for localization you can check /apollo/data/log/localization.out file. Similar for other modules. Alternative is to run specific modules in foreground. Inside apollo docker execute `localization.sh start_fe` - then you'll see what it is outputting. Do this for modules that do not stay enabled.

ajithkrishnan, 2018-12-06:
@martins-mozeiko I have turned on LIDAR, IMU, GPS and Telephoto Camera. And I have not changed the car model and map from "LG SVL 1080" and "San Fransisco" in Dreamview.

The modules that constantly keep crashing are "Control" and "Localization"
Here are the contents of control.out : 
E1109 11:20:53.274397 11081 control.cc:143] Control input data failed: No planning msg
F1109 11:20:53.274461 11081 trajectory_analyzer.cc:69] Check failed: trajectory_points_.size() > 0 (0 vs. 0) 
*** Check failure stack trace: ***
    @     0x7f9eed201ec0  google::LogMessage::Fail()
    @     0x7f9eed201e07  google::LogMessage::SendToLog()
    @     0x7f9eed2017de  google::LogMessage::Flush()
    @     0x7f9eed20483b  google::LogMessageFatal::~LogMessageFatal()
    @           0x70e350  apollo::control::TrajectoryAnalyzer::QueryMatchedPathPoint()
    @           0x691ff4  apollo::control::LonController::ComputeLongitudinalErrors()
    @           0x691010  apollo::control::LonController::ComputeControlCommand()
    @           0x665cf5  apollo::control::ControllerAgent::ComputeControlCommand()
    @           0x41aaca  apollo::control::Control::ProduceControlCommand()
    @           0x41b08a  apollo::control::Control::OnTimer()
    @           0x42f442  boost::_mfi::mf1<>::operator()()
    @           0x42f124  boost::_bi::list2<>::operator()<>()
    @           0x42ed10  boost::_bi::bind_t<>::operator()<>()
    @           0x42e624  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @     0x7f9eef106780  ros::TimerManager<>::TimerQueueCallback::call()
    @     0x7f9eef123ad6  ros::CallbackQueue::callOneCB()
    @     0x7f9eef1242a3  ros::CallbackQueue::callAvailable()
    @     0x7f9eef16ee35  ros::SingleThreadedSpinner::spin()
    @     0x7f9eef1537db  ros::spin()
    @           0x430085  apollo::common::ApolloApp::Spin()
    @           0x418142  main
    @     0x7f9eebdb1f45  __libc_start_main
    @           0x417fb3  (unknown)

and localization.out:
E1109 11:20:45.012897 10945 rtk_localization.cc:85] GPS message buffer is empty.
E1109 11:20:45.022411 10945 rtk_localization.cc:85] GPS message buffer is empty.
...
E1109 11:20:45.078156 10945 rtk_localization.cc:274] [PrepareLocalizationMsg]: time stamp of GPS[1.54176e+09] is different from timestamp of IMU[1.54176e+09]
E1109 11:20:45.088634 10945 rtk_localization.cc:274] [PrepareLocalizationMsg]: time stamp of GPS[1.54176e+09] is different from timestamp of IMU[1.54176e+09]
...
/apollo/bazel-bin/modules/localization/localization: symbol lookup error: /apollo/bazel-bin/modules/localization/localization: undefined symbol: _ZN6apollo12localization3msf12LidarLocator7ComputeEddddddd

I looked for similar error on the official Apollo repo, but i couldn't find a fix.

PS: After the most recent git pull, I get the following error when trying to run the simulation from Menu.unity

/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_library/src/rosbridge_library/protocol.py:381 log() [ERROR] [1544109615.343697]: [Client 0] advertise: Unable to load the manifest for package lgsvl_msgs. Caused by: lgsvl_msgs
ROS path [0]=/home/tmp/ros/share/ros
ROS path [1]=/apollo/ros_pkgs/src
ROS path [2]=/home/tmp/ros/share
ROS path [3]=/home/tmp/ros/stacks

Any advice as to how i can proceed?

martins-mozeiko, 2018-12-06:
You can ignore error message in rosbridge - it simply says that it cannot find .msg file for one of messages it wants to publish. So it will skip this message. It is harmless error message.

Control seems to be crashing because route is not set. That is expected, we see the same behavior from Apollo.

As for localization - it seems to crash because some symbols are missing in its binary. I suggest doing full rebuild - clean bazel workspace (`apollo.sh clean`) and do the build again.

ajithkrishnan, 2019-01-07:
Thanks. This worked.

ajithkrishnan, 2019-01-07:
The full rebuild worked. Thanks.

closed 2019-01-07 by ajithkrishnan

hyunjaecho94, 2019-02-22:
I am having a similar issue. After rebuilding, Edit Routing on Dreamview is still grayed out. "Please send car initial position and map data" message appears, but unlike ajithkrishnan, I have never been able to access Route Editing.. I am getting a rosbridge connection, and its logs are outputting "unable to load the manifest for package pb_msgs. Caused by :pb_msgs". I have turned on GPS and IMU sensors in simulator, so I am thinking that localization is not functioning correctly. What could be the issue? Thank you.