lgsvl/simulator #44: ROS Visualization on local system

ajithkrishnan, 2018-12-05:
Hi,
I am able to read the sensor data on the ROS topics inside the docker container (eg. rostopic echo /apollo/sensor/velodyne64/compensator/PointCloud2). I am also able to view these topics when i execute "rostopic list" on my local computer. However no data is being received in the second case. 
Executing ifconfig in the docker container tells me the ip address is the same as my local computer.
 
I would like to use the sensor data to visualize in Rviz and later process it.

ajithkrishnan, 2018-12-06:
@andytolst This works. Thank You.

closed 2018-12-06 by ajithkrishnan

wangguanbei, 2019-09-16:
@andytolst Sorry, I want to know the position to put
 `-e ROS_MASTER_URI=http://172.17.0.1:11311
-e ROS_IP=172.17.0.1`
to dev_start.sh, can you tell me?

martins-mozeiko, 2019-09-16:
Those are arguments for docker command, so put them anywhere in docker arguments (next to other `-e` args): https://github.com/lgsvl/apollo-3.0/blob/simulator/docker/scripts/dev_start.sh#L265