lgsvl/simulator #47: MD Map Export Does Not Work

taylorsrichard, 2018-12-12:
Hi, I am using the project as is without any changes from the latest merge and I am having an issue exporting the SanFrancisco HD map. Whenever I click "Export HD Map", I get a console error saying that "Some lane segments neighbor relationships are wrong, map generation aborts", followed by "One problematic lane was detected". Can you explain what is causing this issue and suggest a solution? 
Thank you.

hadiTab, 2018-12-12:
@taylorsrichard we seem to have made some mistakes in the HD map annotation we pushed to github. We are working on resolving it. Thanks for pointing it out.

taylorsrichard, 2018-12-13:
@hadiTab Thanks for the prompt response. Glad to know you're working on it.

closed 2019-06-13 by martins-mozeiko