lgsvl/simulator #48: Possible to import real world HD maps into simulator?

taylorsrichard, 2018-12-13:
Hi. I am very interested in how your simulator can convert HD maps generated in Unity into the format used by Apollo. I was just wondering if it would be possible to create a map in unity based off of information from a real world HD map provided from a third party and then export in Apollo format? Is that how you obtained the HD map data for the San Fran map?

martins-mozeiko, 2018-12-13:
Hello.

Currently the workflow is in the opposite way. We first create map in Unity - place all the roads, buildings, other 3d objects. Then do the annotation inside Unity, place lanes, curbs, traffic light locations. And then generate Apollo format map from these annotations automatically.

We have talked about doing it the way you describe - by importing Apollo format map into Unity. But not sure when exactly we would work on this yet.

daohu527, 2019-07-26:
#199

martins-mozeiko, 2019-08-19:
With new simulator [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) release we have implemented import from [lanelet2](https://github.com/fzi-forschungszentrum-informatik/Lanelet2) format. After importing map you should be able to export it to Apollo HD format.

We are working on supporting also OpenDrive and Apollo HD map imports.
If you have any more questions feel free to open new issue.

closed 2019-08-19 by martins-mozeiko