lgsvl/simulator #49: SUV Robot Candidate Missing Functionality

moenasrallah, 2018-12-16:
SF_rigged-apollo robot candidate is missing toggle/menu functionality.
The toggle buttons do not work and enabling sensors have to be toggled manually by the scripts in the Unity inspector
[image]

I would use the default red car, but the suv is nice to have to match our company's cars :)
Any suggested way to modify it myself?

Thank you

martins-mozeiko, 2018-12-18:
If I understand correctly, you added new car prefab to scene and want to run it directly from scene, not through main menu? If so, then check SingleRosConnection object - its property "Robot Setup" should be set to car object that is in scene. The menu will be setup only for this instance.

If that is not the situation you have, then please check the error log. For editor it is in "Console" window, for editor it is Player.log file (location depends on OS: https://docs.unity3d.com/Manual/LogFiles.html#Player). It should have some kind of error if UI does not show up.

EricBoiseLGSVL, 2018-12-19:
The DashUIManager is still in development and isn't set to work when starting in SF.scene.  It is created in menu scene after loading the map selected (MenuScript.cs).  Once the system is finished, DashUI will launch in any scene and remove the single ros connection dependency to make development easier.

moenasrallah, 2018-12-19:
@EricBoiseLGSVL thank you for that info

closed 2018-12-19 by moenasrallah

moenasrallah, 2018-12-19:
Update:

1) Added custom scene to the map selector of the main menu
2) Added custom vehicle prefab to the vehicle selector of the main menu
3) Now after reading @EricBoiseLGSVL 's comment, the DashUIManager shows up and works successfully with the custom car :)