lgsvl/simulator #50: wher is lgsvl-msgs

hakuturu583, 2018-12-18:
I tested your binary. (https://github.com/lgsvl/simulator/releases)
And I found that there are no lgsvl_msgs in my PC.
So your simulator failed to send ground truth to Autoware.

hakuturu583, 2018-12-18:
[image]

daviduhm, 2018-12-19:
@hakuturu583 We've just published a new repo for message definitions: [lgsvl_msgs](https://github.com/lgsvl/lgsvl_msgs). Please place this ROS package under /autoware/ros/src/msgs/ and build to subscribe ground truth messages in Autoware. Please let me know if you have any questions regarding the message types.

closed 2018-12-19 by hakuturu583

yangydavid, 2019-01-09:
Hi @daviduhm The same error happens when I use your simulator with Apollo, could you let me know which directory in Apollo should I put this package? Thank you!

asimonov-hitachi, 2019-01-09:
thanks @daviduhm !

I think this should be mentioned on setup instructions. Or alternatively added to lgsvl_autoware clone.

daviduhm, 2019-01-10:
@yangydavid Yes, you can use the same [lgsvl_msgs](https://github.com/lgsvl/lgsvl_msgs) repo for Apollo as follows:
1. Put lgsvl_msgs package in [/apollo/ros_pkgs/src/](https://github.com/lgsvl/apollo/tree/simulator/ros_pkgs/src)
2. Build the package by running `catkin_make` in /apollo/ros_pkgs/
3. Then, `source /apollo/ros_pkgs/devel/setup.bash`

Once you build and source the lgsvl_msgs package successfully, the error message should be gone and you can now rostopic echo the ground truth messages via rosbridge connected to the simulator when you enable the ground truth feature.

daviduhm, 2019-01-10:
@asimonov-hitachi The lgsvl_msgs package will be included in our forked repositories for Apollo and Autoware in next push. Thanks for pointing it out!

daviduhm, 2019-01-10:
@asimonov-hitachi The lgsvl_msgs package will be included in our forked repositories of Apollo and Autoware in next push. Thanks for pointing it out!

yangydavid, 2019-01-10:
@daviduhm  Thanks! just got it to work :)