lgsvl/simulator #51: Autoware & LG SVL :  Failed to find match for field 'intensity'.

MouriNaruto, 2018-12-19:
When I click the "Localization" button in Autoware and found this message on one of these consoles.

Environment:
- OS: Ubuntu 16.04.5
- LG SVL Simulator: https://github.com/lgsvl/simulator/releases/tag/2018.12-rc1
- Autoware in Docker: https://github.com/lgsvl/Autoware/tree/lgsvl_develop

[image]

How to address it? I am looking forward to hearing from you soon. Thanks.

Mouri.

hadiTab, 2018-12-19:
@MouriNaruto that is a warning message from PCL (point cloud library) which autoware uses. The warning should not affect anything. If I recall correctly it is because PCL does not expect a field named "intensity" in the data. Are you having any issues with the localization or lidar other than that warning message? i.e. does the localization look correct in rviz?

MouriNaruto, 2018-12-19:
@hadiTab I read the guide. But I wait for several minutes and didn't find the time to display to the right of "Ref" of "Localization" in the Autoware Runtime Manager.

MouriNaruto, 2018-12-20:
[image]
@hadiTab  I can see the similar screenshot as yours. But the car in the simulator didn't move and find a terminal window says:

```
PlannerH -> Plan (A) Failed, Trying Plan (B).
Can't Generate Global Path for Start (X:214.665590521, Y:201.860796092, Z:9.998589, A:-3.14066762909
Lon:0, Lat:0, Alt:0, Dir:0
) and Goal (X:107.247931616, Y:212.341751099, Z:0, A:2.55358978788
Lon:0, Lat:0, Alt:0, Dir:0
)
```

I wonder to know why. Thanks.

Mouri.

hadiTab, 2018-12-20:
@MouriNaruto that message is saying that the global planner is not able to find a route from point A (the car location) to point B. If you look at the map closely you will notice that all destinations are not achievable given the direction of the streets on the vector map (see the small pink arrows on the lane centerlines in the image you've attached). Also, Autoware does not really support lane change, so any destination you choose will need to reachable from the lane the car is currently in. It's simpler to start out trying a destination that is directly ahead of the car in the same lane. Once you get the hang of it you will realize which destinations are reachable.

Also, note that once a destination is rejected (as seen in your image above) you will need to relaunch the "Mission Planner" (click the button twice) to be able to try another destination.

MouriNaruto, 2018-12-23:
@hadiTab Thanks, I can see the car was running. But there are some problems.

1. No matter the target point I choose, the car will always stop at here. Even if my target point is very close, the car can not stop at the target point. The car wasn't stopping until to here.
[image]
2. `Autoware does not really support lane change`  Really? Autoware? Not the simulator? Does Apollo support it?

hadiTab, 2018-12-23:
1. I've observed this problem as well. The problem seems to be related to the pure_pursuit node in Autoware. If you open the settings for the node (it's under the 'compute' tab in the Autoware GUI) there are two modes for running the node: 1. waypoint 2. dialog. In waypoint mode the ego vehicles velocity and look ahead distance are automatically configured wheras in dialog mode they are manually configured. If you switch to dialog mode and play around with the velocity and the look ahead distance you should probably be able to get it to make the turn. However, this could make the vehicle drive very slowly at other parts of the map.

2. Yes, unfortunately lane changing is not really supported in Autoware and it is disabled by default. It is possible to enable it by changing a few lines of code, but in our experience it does not really work when enabled (it plans unfeasible routes). In general the simulator does not care about the implementation of planning and routing algorithms in the autonomous stack. It only takes the control commands from the autonomous stack and follows directions. The lane change functionality does work in Apollo, and at the moment Apollo works a bit better with the simulator.

MouriNaruto, 2018-12-24:
@hadiTab Thank you very much.

1. I am looking forward to hearing the fix from this project.
2. Got it.

Finally, I will close the issue.

Mouri.

closed 2018-12-24 by MouriNaruto

lrodrig3, 2019-01-09:
I was having the same problem, but once I reload the mission planning, I'm getting this error

`[ERROR] [1547025039.319993704]: Client [/rviz_1547024734375863050] wants topic /local_trajectories to have datatype/md5sum [visualization_msgs/MarkerArray/d155b9ce5188fbaf89745847fd5882d7], but our version has [autoware_msgs/LaneArray/ca01c32675aeaa32240193d4d3f7e07a]. Dropping connection.`

zerolover, 2021-01-29:
@MouriNaruto 
 Failed to find match for field 'intensity'. is from [pcl::fromROSMsg(*input, scan](https://github.com/Autoware-AI/core_perception/blob/411330c5bb0eb5a963393f869c19d7815bb2dafc/points_downsampler/nodes/voxel_grid_filter/voxel_grid_filter.cpp#L60))
PointCloud `scan` in PCL has format : PointXYZI (which is float * 4)
But `input` is X,Y,Z,I,timestamp (float,float,float,uint8,...), see [link](https://github.com/lgsvl/simulator/blob/8b02c852c0642347f0b05125c726ceb09bfa0b26/Assets/Scripts/Bridge/Ros/RosPointCloudWriter.cs#L92)
So fromROSMsg cannot convert msg correctly, intensity data in scan are all 0.