lgsvl/simulator #53: simulator cannot connect to rosbrige

smilingpaul, 2018-12-26:
Hi all,
My simulator cannot connect to rosbrige, in apollo docker.
tail  ~/.config/unity3d/Editor.log 
show the following err:
12/26/2018 7:05:06 PM|Fatal|WebSocket.connect|An address incompatible with the requested protocol was used
I follow the Tutorial, 
http://wiki.ros.org/roslibjs/Tutorials/BasicRosFunctionality
it get connected, but with errors:
/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_library/src/rosbridge_library/protocol.py:381 log() [ERROR] [1545815301.545071]: [Client 3] [id: call_service:/rosapi/get_param_names:5] call_service InvalidServiceException: Service /rosapi/get_param_names does not exist

how can I resolve it?  It seems that simulator C# code using websocket-sharp.dll. Any help appreciate. thanks.

hadiTab, 2018-12-26:
Are you running the simulator in the Unity editor or as a binary? If you are using the editor please check if there are any exceptions there.

As for the tutorial you've posted, is the error message you posted only coming up when you follow the tutorial steps? i.e. if you just launch the bridge in the container without following the tutorial do you still get that error? Does it appear once a certain message is being published?

smilingpaul, 2018-12-28:
@hadiTab thanks for your response. 
I change bridge address "localhost"   to "127.0.0.1", on the FreeRoaming page configuration, it got connected.