lgsvl/simulator #54: simulator localhost disconnect when launched

overlapjho, 2018-12-26:
I am not able to run the car following the tutorial for quick-start everytime I connect to localhost it disconnect.
[image]
[image]

hadiTab, 2019-01-04:
@overlapjho sorry for the late reply. I have not been able to recreate this issue so it's a bit difficult for me to debug it. Can you walk me through the steps you take to get there?
Also, are you using the latest simulator and autoware repo?
Thanks.

overlapjho, 2019-01-05:
@hadiTab  Ya, I'm using the latest Sim and Autoware repo here is the steps that I did:

(1) With autoware

```
1. cd autoware/ros
2. ./run
3. Clicked map and sensing
4. Open simulator at lowest quality settings
5. Clicked free roaming 
6. Select SanFrancisco map then EP_rigged_apollo (I see Connected status)
7. Run
8. Check on upper left if rosbridge connects but it shows "Connecting"
9. Press F1 to verify and saw on available ros bridge: localhost (Disconnected)
10. Checked the terminal for error messages and it connects then disconnects per sec
```

(2) Terminal
```
1. Open simulator at lowest quality settings
2. Clicked free roaming 
3. Select SanFrancisco map then EP_rigged_apollo
4. launch roslaunch rosbridge_server rosbridge_websocket.launch
5. (I see Connected status) then Run 
6. Check on upper left if rosbridge connects but it shows "Connecting"
7. Press F1 to verify and saw on available ros bridge: localhost (Disconnected)
8. Checked the terminal for error messages and it connects then disconnects per sec

hadiTab, 2019-01-07:
@overlapjho in step 6 you should be choosing XE_rigged_autoware as the vehicle, otherwise there will be some errors. Please check and make sure this isn't causing the issue.

Otherwise, I noticed that you are running an autoware installation on your own machine, another suggestion would be to uninstall rosbridge from your machine and use our fork of rosbridge available [here](https://github.com/lgsvl/rosbridge_suite) which should work out of the box. If not, I believe it will print more errors than the default rosbridge.

overlapjho, 2019-01-08:
@hadiTab Do you have any instruction how to install the forked rosbridge_suite? I can't find any documentation

hadiTab, 2019-01-08:
@overlapjho you basically need to place it in a ros workspace and build it with catkin. The following steps should do the trick:
- Create a new folder for the rosbridge package:
`mkdir -p rosbridge_pkg/src && cd rosbridge_pkg/src`

- Clone the rosbridge repo:
`git clone git@github.com:lgsvl/rosbridge_suite.git`

- Build:
`cd ..`
`catkin_make`

- Make sure to source the setup file in each terminal window you want to run rosbridge in.
`source devel/setup.bash`
`roslaunch rosbridge_server` rosbridge_websocket.launch`

Make sure to comment out the line in the sensing.launch file for Autoware which is related to launching the bridge.

overlapjho, 2019-01-08:
Great! It resolve the first issue this error shows though
```
[ERROR] [1546908645.452988]: [Client 0] subscribe: Unable to load the manifest for package autoware_msgs. Caused by: autoware_msgs
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/overlapjho/rosbridge_pkg/src
ROS path [2]=/opt/ros/kinetic/share
[ERROR] [1546908645.458024]: [Client 0] advertise: Unable to load the manifest for package lgsvl_msgs. Caused by: lgsvl_msgs
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/overlapjho/rosbridge_pkg/src
ROS path [2]=/opt/ros/kinetic/share```

Can I safely ignore this or is there any workaround to solve it?

closed 2019-01-08 by overlapjho

hadiTab, 2019-01-08:
@overlapjho great! You should be able to safely ignore those messages.