lgsvl/simulator #55: failed to run in Windows

hakuturu583, 2018-12-27:
I built your sim in Windows and Unity, the build process was succedd.
But when I launch your sim, I found that the display is strange.
Please tell me the some instructions.

[image]

hadiTab, 2018-12-27:
It looks like an lfs issue. Can you make sure that git lfs is installed on your windows machine?

hakuturu583, 2018-12-28:
I add git lfs module and build your simulator, but I failed to load map.

[image]

hakuturu583, 2018-12-28:
All map file was loaded when I build scenes in Unity Editor.
[image]

hakuturu583, 2018-12-28:
I failed to find maps when I select build & run

EricBoiseLGSVL, 2018-12-28:
Did you build from the command line?  There are parameters that need implemented.  This will happen if you build from Editor.

Link https://github.com/lgsvl/simulator/blob/master/Docs/build-instructions.md

hakuturu583, 2019-01-07:
I build from Unity Editor

hakuturu583, 2019-01-07:
But, all scenes are found when I build simulator when I click right arrow mark on the top of the Unity Editor.
[image]

martins-mozeiko, 2019-01-07:
To build player you cannot use Unity default "Build" option in Editor to create binary builds. Our simulator requires extra build steps that will correctly package map assets. You need to build on commandline with command written in our build instructions: https://github.com/lgsvl/simulator/blob/master/Docs/build-instructions.md
Change `-buildTarget Linux64` to `-buildTarget Windows64` when building on Windows.