lgsvl/simulator #56: failed to connect to Windows Simulation machine.

hakuturu583, 2018-12-28:
I tested your simulator in Windows and run ROS in Linux.
I checked communication confirmation was succeed, and try to run rosbridge_server.
But it was failed and I faced error logs below.

[image]

hadiTab, 2018-12-28:
@hakuturu583 can you provide more details on what steps you are going through? Thanks.

hakuturu583, 2019-01-07:
@zelenkovsky @hadiTab I setted IP v4 adress like below, but the rosbridge was failed to connect.
[image]

hakuturu583, 2019-01-07:
I succeed to connect now.

closed 2019-01-07 by hakuturu583