lgsvl/simulator #58: The error messages in lgsvl simulator with Autoware.

oiz5201618, 2019-01-01:
Hi, thanks in advance for any reply.
If this issue is not suitable for this repository, please inform and close it, thanks!

When I try to run Autoware with lgsvl_simulator, it occur some error as below.
My Autoware runs with kinetic version in the docker container and all installations exactly follow the guide.

1. The **Localization** don't display the time at the right of "Ref", and the error message repeatedly shows up `Failed to find match for field 'intensity'.`
[image]

2. Next, the **Detection** also repeatedly warns `[ INFO] [1546326933.338934217]: wayarea layer not contained in the OccupancyGrid`, but this is only the **INFO** message.
I'm not sure whether this message affects the system or not.
[image]

3. Final, once the **Motion Planning** launches, the `config_waypoint_follower_rostopic-6` process has died.
[image]

After these three error messages, my Autoware still can operate the sensor fusing in **Detection** and plan the path in **Mission Planning**, but the vehicle just stucks.

[image]

hadiTab, 2019-01-02:
@oiz5201618 it seems like you have autoware setup pretty well with the simulator. I've addressed the issues bellow.

1. This is related to the PCL library not expecting to see an intensity field in the pointcloud map I believe. It should not affect anything.

2. This one should also have no effect. As far as I can recall it is Autoware detection tries to match the obstacles to the wayarea (the drive-able areas) and it is failing to do so because the HD map does not contain that information. It is an optional functionality though and detection should work without it.

3. Unfortunately, I am not entirely sure what the reason for this one is. It happens sometimes for me as well. It sometimes happens on some computers when the waypoint follower node is launched through the launch file (the node is dying immediately after being launched). It does not happen when the node is launched manually. To launch it manually follow these steps:
- Switch over to the 'Computing' tab in the runtime manager.
- Scroll down and find 'pure_pursuit' and 'twist_filter' in the right side column (probably under the title waypoint follower) and check the box in front of both of them.
- The car will likely start moving now. The motion may not be smooth though. You can tweek the motion parameters by clicking on the button to the right of 'pure_pursuit' and changing the configuration parameters.

Let me know if this helps.

oiz5201618, 2019-01-04:
@hadiTab thanks for your detailed reply.
The manual launch can avoid the process died problem, but the vehicle still stucks.

And I encountered another problem as below.
* My sensor fusing just keeps rotating in rviz, but no error messages show up.
---
[image]
---
[image]
---
[image]
---

The following is my device infomation. Is my computing power not enough?
* CPU: Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz with 4 cores 8 threads.
* RAM: 16GB
* GPU: GeForce GTX 1060 6GB

oiz5201618, 2019-01-04:
@hadiTab 
I solved the sensor data rotating problem by restarting my simulator, it seems some configuration went wrong, but the vehicle still cannot move.

[image]

When I open rviz, the error message shows up.
[image]

hadiTab, 2019-01-04:
@oiz5201618 I'm not sure why you are seeing that error, but it should not affect the vehicle moving. It seems like some of the message definitions in Autoware have changed and do not match what rviz is expecting, but that would only affect the rviz visualization.

Try the following:
1. Launch everything, switch over to the 'Computing' tab and launch pure_pursuit and twist_filter.
2. Open the settings menu for pure_pursuit and switch the mode to 'dialog'.
3. Increase the speed to about 15 and the lookahead distance to about 5. If the vehicle does not move try tuning these two variables. The effect should be immediate, i.e. as you move the slider the variables are changed instantaneously and you do not need to click OK for it to take affect. Play with these two variables until the vehicle starts to move.

oiz5201618, 2019-01-06:
If I send the control command by `rostopic pub`, the vehicle moves.
`$ rostopic pub /twist_cmd geometry_msgs/TwistStamped ${Arg}`

But if I give the goal by the same method, the vehicle still stucks.
`$ rostopic pub /move_base_simple/goal geometry_msgs/PoseStamped ${Arg}`

It seems the motion_planner doesn't work properly.

oiz5201618, 2019-01-08:
Hi, @hadiTab 
Thank you very much for your kindly reply.

I solved the problem by enabling the **astar_planner** function in the 'Computing' tab, which is also launched in the `my_motion_planning.launch` file. In fact, we also can switch the motion planner to the other one (e.g. lattice_planner), the vehicle will move properly.

[image]

So the problem is that **astar/lattice_planner** which provides a reference path for **pure_persuit** dosen't be launched, and I think I misunderstood the steps you mentioned.

But if I enable traffic in the Unity, the vehicle collides with the vehicles ahead instead of avoiding or stoping. It seems the avoidance function doesn't work despite I enable it in the astar_planner.

[image]

closed 2019-08-20 by hadiTab