lgsvl/simulator #59: Setting custom robotCandidate in RosRobots.cs

ajithkrishnan, 2019-01-02:
Hi,
I have created a custom vehicle based on the XE-Rigged-Apollo vehicle which runs with the apollo stack. Until now i have built the repo using Unity and run the simulation after adding my vehicle to the robotCandidates list in Menu.unity directly in the editor.

[image]

However i would like to build using the following command -
"/opt/Unity/Editor/Unity \
    -batchmode \
    -nographics \
    -silent-crashes \
    -quit \
    -buildDestination ./build/simulator \
    -buildTarget Linux64 \
    -executeMethod BuildScript.Build \
    -projectPath . \
    -logFile /dev/stdout
"

which runs the simulation directly (bypassing the Editor). Due to this i am not able to add the said vehicle to the list of robots in the drop down menu. Is there a way to add my vehicle in RosRobots.cs so that it appears along with the other bots?

[image]

I have already copied my vehicle prefab to the Assets/Prefabs/RobotCandidates folder.

Thank you.

ajithkrishnan, 2019-01-02:
Solved this. I added the vehicle prefab to the robotCandidates list and saved Menu.unity scene. Then rebuilt the simulation using 
"/opt/Unity/Editor/Unity
-batchmode
-nographics
-silent-crashes
-quit
-buildDestination ./build/simulator
-buildTarget Linux64
-executeMethod BuildScript.Build
-projectPath .
-logFile /dev/stdout
"
And now it shows up.

EricBoiseLGSVL, 2019-01-02:
Glad you were able to add the prefab.  Since MenuScript.cs is not attached to a prefab in Menu.scene, it requires you to save the scene to serialize changes.

ajithkrishnan, 2019-01-03:
@EricBoiseLGSVL Thanks

closed 2019-01-03 by ajithkrishnan