lgsvl/simulator #63: How to get vehicle state information

sygin123, 2019-01-07:
Hi guys, thanks for your amazing job! I could run autoware demo following the instruction. However, I can't find  any document about how to parse vehicle's velocity, steer angle, etc. The topic /vehicle_status seems to be empty. Is there any way to get that ? thanks a lot.

martins-mozeiko, 2019-01-07:
Hello!

Are you asking about ground truth values? Or are you interested in information that autoware publishes to the simulator to steer the car?

Ground truth values for information you are asking is not yet implemented. We are planning to offer functionality that will allow to extract various kind of information out form simulator, but that will come later.

As for what autoware publishes to steer the car, this is available in `/vehicle_cmd` topic. You can see how simulator is using information from autoware in [VehicleInputController.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/VehicleInputController.cs#L207) file.

sygin123, 2019-01-08:
Hi @martins-mozeiko ,

Appreciated your reply so quickly!  I am asking exactly for the ground truth values. Is that means the current motion planner in autoware does not take any feedback from simulator into consideration?  

Looking forward to more interface information released !  Can you share something about the roadmap?

By the way, the car will always stuck in the 90 degree turn, it is so strange.
[image]

hadiTab, 2019-01-14:
@sygin123 Autoware has not implemented any low level controllers. Autoware sends planner outputs in the form of twist velocity commands to a third party controller which uses the CAN bus for communication and all of it's internals (including feedback) is invisible to us.

If you are interested in working on low level controllers with the simulator it may be best to use Apollo which uses a low level controller. If you run our simulator with Apollo vehicle selected the simulator will publish feedback for steering angle, throttle percentage, brake percentage, etc. However, all of these are published on a protobuf message so you would need to use Apollo.

For Autoware, we want to provide the feedback as well. Even-though Autware doesn't use it for low-level control. The information would be useful for other purposes. I do not have an exact timeline for it though.

closed 2020-04-07 by EricBoiseLGSVL