lgsvl/simulator #64: Customize Traffic Scenarios

yangydavid, 2019-01-08:
Dear Developers, 

I am wondering if we are able to put a seed into the traffic so that it follows exactly same pattern every time we enable it?

EricBoiseLGSVL, 2019-01-14:
Hello @yangydavid 

We are working on a scenario system but unfortunately, it isn't ready yet.  Hopefully we can accomplish this soon.

-Eric

martins-mozeiko, 2019-08-19:
With new simulator [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) release we have a random seed as input to simulation. Same simulation will generate exactly same NPC vehicle movement.

If you have any issues with this new feature, please open a new issue.

closed 2019-08-19 by martins-mozeiko

martins-mozeiko, 2019-08-19:
@yangydavid We have released new [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) version or our simulator which includes much simpler way to customize or add additional sensors to vehicle. See the documentation [here](https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-change-the-configuration-of-a-vehicle) and [here](https://www.lgsvlsimulator.com/docs/sensor-json-options/).

If you have any more questions fee free to open new issue.