lgsvl/simulator #68: how to define a stereo camera for Autoware?

asimonov-hitachi, 2019-01-10:
hi!
Great job with the simulator and integration with Autoware!

I can successfully build the simulator in Unity. I can change Lidar settings and see the changes in simulator.

What I am struggling with is how to define a stereo camera and then have it appear in autoware/ROS.
Can you point me in the right direction?
I do not have any experience with Unity, but can follow instructions :)

Also, is there a high level overview of how data gets from the simulator to ROS1 container? Is there ROS2 involved? This links to previous question as I basically wanna modify different parts to do what I need.

martins-mozeiko, 2019-01-14:
Hello!

If you want an extra camera on Autoware vehicle you need to do following steps:

1) Open SanFrancisco scene
2) Drag in Autoware prefab into scene: 
[image]

3) Duplicate (copy & paste) CaptureCamera object inside XE_Rigged-autoware scene object:
[image]

4) Select new "Capture Camera (1)" object and adjust its position/orientation in 3d view:

[image]

5) Change properties of new "CaptureCamera (1)" object - change `TopicName` to different topic, and `Sensor Name` for UI.

[image]

6) Open new Inspector tab (1), in one inspector select XE_Rigged-autoware game object (2) and press lock icon to lock it (2), then select new "CaptureCamera (1)" game object in second inspector (3). Now increase number for `NeedsBridge` property in XE_Rigged-autoware component and drag & drop (4) `Video To Ros` script from camera into `Needs Bridge` list.

[image]

7) Press "Apply" at top of inspector of XE_Rigged-autoware game object to save it.

Do not save the SanFrancisco scene, as you don't want autoware game object there. It will be added dynamically.

Now if you run simulator from Main Menu, the autoware vehicle should publish to camera topics.

Let me know if you have more questions on this.

asimonov-hitachi, 2019-01-31:
thank you! i will try this later

asimonov-hitachi, 2019-03-29:
That worked! Thanks a lot!

asimonov-hitachi, 2019-03-29:
actually, if I keep using the config.yaml (even if I delete     enable_main_camera : true from XE_Rigged-autoware) the simulator does not start - i just see black screen forever.
Can it work with config.yaml?

martins-mozeiko, 2019-03-29:
This probably means that there is some exception happening that prevents simulator for running. 

You can check what exception is happening inside log file in `/.config/unity3d/LG Silicon Valley Lab/Automotive Simulator/Player.log` location on Linux, or `%APPDATA%\..\LocalLow\LG Silicon Valley Lab\Automotive Simulator\output_log.txt` location on Windows.

closed 2020-04-07 by EricBoiseLGSVL