lgsvl/simulator #70: speed of data transfer from simulator to ROSbridge/websocket

asimonov-hitachi, 2019-01-11:
hi, I am using the simulator with Autoware and Apollo and seeing that some image topics arrive at ROS container topic at around 5Hz. The simulator screen sets image rate at 15Hz.

I think it is essentially this issue: https://github.com/RobotWebTools/rosbridge_suite/issues/344

Any ideas how this can be worked around?

martins-mozeiko, 2019-01-14:
Hello!

Are you using our modified rosbridge from https://github.com/lgsvl/rosbridge_suite ? It has few modifications to make transfers a bit faster.

That said, we are aware that current bridge implementation may be slow for larger data depending on your PC and we are thinking how to improve.

Currently you could try reducing jpeg quality VideoToROS.cs script - change JpegQuality property on camera objects (CaptureCamera and TelephotoCamera for apollo) to lower number. Try something like 30, this will increase jpeg compression and give you smaller size to send to ROS.

asimonov-hitachi, 2019-01-31:
I use whatever is in provided autoware-docker (https://github.com/lgsvl/Autoware)

asimonov-hitachi, 2019-03-29:
just managed to run with lgsvl's version of rosbridge_suite. Using XE_rigged-autoware model in simulator.
It does not make any difference.
In summary, running only rosbridge inside the ROS container and simulator on Linux, I see:
LIDAR data is coming to ROS at 10Hz
compressed/image data is coming to ROS at 4Hz, but FPS indicator in simulator is saying 8 Hz.

On another machine I also see published image topic inside ROS at about half the hz of FPS in the simulator.

martins-mozeiko, 2019-03-29:
There are two things that makes image sending slow.

One is framerate of simulator itself. Due to how GPUs work, the camera image is read back on next frame. So simulator won't be reading and sending images faster than half of your frame rate. That is why you see 4Hz sending rate when you have 8Hz framerate for rendering. This is not related to rosbridge at all. To fix this you need to improve rendering speed of simulator. What CPU and GPU are you using? If you are running it on 4K monitor fullscreen, then try reducing window size to 1080p or so. One thing we recommend is running simualtor on Windows, because Unity and Nvidia drivers for graphics rendering are more optimized on Windows than on Linux. It will probably improve in future on Linux, but other than getting more powerful CPU & GPU won't help much on Linux.

You can determine if rosbridge performance is an issue if rosbridge process is running constantly with 100% CPU usage. That would mean it cannot receive & process data realtime. If process usage is less than 100% then rosbridge is not an issue here.

rukie, 2019-12-18:
Other companies offer "shared memory" between their simulators and ROS. However, this requires running both the simulation and ROS on the same machine.

closed 2020-04-07 by EricBoiseLGSVL