lgsvl/simulator #71: Impressive performance - LIDAR Realism

rukie, 2019-01-11:
I'm thoroughly impressed by your GPU accelerated LIDAR sensor! Great work!

I do like that you provide a ROS PointCloud2, but at high speed this setup doesn't easily allow for fixing point cloud distortion, in a manner that roles back to a real vehicle.

It looks like your distortion is limited to the azimuth change of the sensor over a fixedUpdate(). Adding an estimated vehicle distance traveled to each point could smooth that out. 

It would be nice to recreate some real hardware with ethernet traffic, similar to https://github.com/vwaurich/VelodyneLidarUnitySimulation with your GPU setup. 

The messages could then be read into ROS with the Velodyne driver: 
https://github.com/ros-drivers/velodyne

Again, incredible work!

martins-mozeiko, 2019-01-14:
Hi.

Thanks for the information. We are working on fixing the distortion and improving its performance, this is definitely high on our to-do list.

asimonov-hitachi, 2019-03-29:
I agree that pushing simulated UDP velodyne packets seem to be a better solution. 
Unless you try to limit the interfacing to rosbridge_suite only.

rukie, 2019-04-02:
By modifying this guy's UDP send script with some parallelization, performance significantly increases. Off the bat, its kind of slow.
https://github.com/vwaurich/VelodyneLidarUnitySimulation

atyshka, 2019-07-19:
How does the bandwidth work with LIDAR and rosbridge? I would think that translating from JSON would be a relatively inefficient and create some performance bottlenecks but maybe not.

EricBoiseLGSVL, 2020-04-07:
@rukie @atyshka We are finalizing a UDP solution and will be releasing soon

closed 2020-04-07 by EricBoiseLGSVL

rukie, 2020-04-09:
Exciting news! Thanks for the update.